Ultimaker/Cura #10943: Can't chang diameter of an filament (diameter of an filament not save)

userosos, 2021-11-28:
### Application Version

4.12.1

### Platform

Windows 11

### Printer

Ulti

### Reproduction steps

1. Press ctrl+k
2. Set information/properties/diameter to 1,73 form 1,72.
3. Close the windows
4. Press ctrl+k
5. Can see that information/properties/diameter is 1,72 (in step 2 i set it to 1,73)

### Actual results

diameter filament not save

### Expected results

I want changin diametrs of filaments

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[log.zip](https://github.com/Ultimaker/Cura/files/7612990/log.zip)

fieldOfView, 2021-11-28:
Try using a comma instead of a period, or viceversa.

userosos, 2021-11-28:
> Try using a comma instead of a period, or viceversa.

I cant use dot in the place. Russian locale.

fieldOfView, 2021-11-28:
I see. That'll be the problem then... Cura expects a period, but will only let you enter a comma because of the locale.

fvrmr, 2021-11-29:
Hi @userosos thank you for your report.
I will bring this up with the team. Keep you posted!

userosos, 2021-12-01:
> I see. That'll be the problem then... Cura expects a period, but will only let you enter a comma because of the locale.

I cant set in the place an dot from keyboard. If i press the key i can't see the dot in the place.

userosos, 2021-12-03:
If i not sliced before set the settings it work ok.

fvrmr, 2021-12-07:
I have discussed it with the team and I created a ticket of this.
Devs see CURA-8741

GregValiant, 2024-11-16:
Is this still a problem in current versions of Cura (5.8.0 and up)?  Can this be closed?

closed 2024-12-09 by github-actions[bot]