Ultimaker/Cura #10944: Arrange all models fails to arrange

hapklaar, 2021-11-28:
### Application Version

4.11.0 and 4.12.1

### Platform

WIndows 10

### Printer

Biqu B1

### Reproduction steps

1. Start Cura
2. Drag foot.stl (see attached file) to Cura
3. Multiply selected by 2 to get 3 copies of the part

### Actual results

Cura places the 2 extra parts, with one of them outside the buildplate with an error

"Unable to find a location within the build volume for all objects"

Tried to arrange again by selecting all models, then ctrl-r. Getting same error:

[image]

Manually arranged models without issue. The build plate can easily support 4 copies at the same time:

[image]

### Expected results

1. For Cura to place all 3 or 4 copies on the build plate without giving an error when multiplying
2. For Cura to arrange models properly based on the available build volume

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Foot.zip](https://github.com/Ultimaker/Cura/files/7613368/Foot.zip
[BB1_Foot-project.zip](https://github.com/Ultimaker/Cura/files/7613388/BB1_Foot-project.zip)
)

fieldOfView, 2021-11-28:
It turns out that you are smarter than Cura. Cura is not all that smart, it just follows rules. One of the rules it follows is that when starting to arrange, it will put one model in the center and then arrange the rest around that. This approach does not always work out.

fvrmr, 2021-11-29:
Hi @hapklaar thank you for your report.
This is a duplicate of #9793 there you can read that this is happening because we are depended of the library we use for arranging models.

I will close this ticket as a duplicate, I hope you understand.

closed 2021-11-29 by fvrmr