Ultimaker/Cura #10945: In preview, playing the layer, Infill print 3 times

physicsimulation, 2021-11-28:
### Application Version

Arachne Engine Beta 2

### Platform

Windows 10

### Printer

Creality Ender-3

### Reproduction steps

1) Create an object including Infill.
2) In preview, play the layer.

### Actual results

Print infill 3 times

### Expected results

Print Infill 1 times

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Nothing else to say

fvrmr, 2021-11-29:
Hi @physicsimulation thank you for your report.
Could you share a project file? (file - save project)
With a project file it is easier for us to reproduce your bug.

physicsimulation, 2021-11-29:
1) Import the "benchy" object
2) Preview "layer 15" (or any layer with infill) where you got infill, skin and wall
3) Play Sequence : Infill, Walls, Skin
4) where Infill with extra wall
5) where Wall (outer+ inner)
6) Skin with Extra
I uninstalled Arachne to "fix" this bug. Sorry.

fvrmr, 2021-12-07:
Hi @physicsimulation thank you but I can't reproduce your issue with these steps. To discuss this with the team I need a project file. So I will leave this issue open and wait if other people have a project file to reproduce this.

GregValiant, 2024-11-16:
Is this still a problem in current versions of Cura (5.8.0 and up)?  Can this be closed?

closed 2024-12-09 by github-actions[bot]