Ultimaker/Cura #10949: Cura 4.12.1 crashes on exit MacOS 12.0.1 (Monterey)

SmithyAT, 2021-11-28:
### Application Version

4.12.1

### Platform

MacOS 12.0.1

### Printer

Ultimaker S5

### Reproduction steps

Quit/Exit Cura normally

### Actual results

Cura crashes every time I want to quit the application with a crash report:

Process:               cura [15891]
Path:                  /Applications/Ultimaker Cura.app/Contents/MacOS/cura
Identifier:            nl.ultimaker.cura
Version:               4.12 (4.12.1)
Code Type:             X86-64 (Translated)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2021-11-28 19:22:15.7682 +0100
OS Version:            macOS 12.0.1 (21A559)
Report Version:        12
Anonymous UUID:        C17FDA89-4EFE-3637-3907-7C547A0E969F

Sleep/Wake UUID:       D09D74ED-FD25-4B71-8F24-4A6E10780B5A

Time Awake Since Boot: 130000 seconds
Time Since Wake:       623 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Codes:       0x0000000000000001, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

### Expected results

Normal termination of the application

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/7614111/cura.log)

fvrmr, 2021-11-29:
Hi @SmithyAT thank you for your bug report. 
We had this bug before but was fixed in 4.11, see: https://github.com/Ultimaker/Cura/issues/9837
What language are you using in Cura? 

I will bring this up with the team. Keep you posted!

SmithyAT, 2021-11-29:
The Cura language is English, the system OS language is German.

Strange, I never noticed the problem with 4.11, not 100% about 4.12, but I think there was no problem. For me it started with 4.12.1

SmithyAT, 2021-11-29:
It is really strange, I wanted now to try some test cases, to maybe see when it happens, but currently the error don't appear anymore. I haven't done any changes or rebooted the Macbook, absolutely nothing.

Will report back when it happens again, sorry for confusion, but it really happened every time I quit Cura and now ... :-)

fvrmr, 2021-11-30:
Thank you for letting me know. If you have any updates I will bring it up with the team again!

ikkez, 2022-04-15:
Same problem here. I have this error for months now every time I exit cura.

[image]

Full log:

```
Thread 0x00007000140d0000 (most recent call first):
  File "/Users/ultimaker/build/env/4.13/install/lib/python3.8/threading.py", line 306 in wait
  File "/Users/ultimaker/build/env/4.13/install/lib/python3.8/site-packages/zeroconf/__init__.py", line 2534 in wait
  File "/Users/ultimaker/build/env/4.13/install/lib/python3.8/site-packages/zeroconf/__init__.py", line 1715 in run
  File "/Users/ultimaker/build/env/4.13/install/lib/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/Users/ultimaker/build/env/4.13/install/lib/python3.8/threading.py", line 890 in _bootstrap

Thread 0x00007000130cd000 (most recent call first):
  File "/Users/ultimaker/build/env/4.13/install/lib/python3.8/site-packages/zeroconf/__init__.py", line 1369 in run
  File "/Users/ultimaker/build/env/4.13/install/lib/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/Users/ultimaker/build/env/4.13/install/lib/python3.8/threading.py", line 890 in _bootstrap

Thread 0x00007000120ca000 (most recent call first):
  File "/Applications/Ultimaker Cura.app/Contents/MacOS/plugins/plugins/USBPrinting/USBPrinterOutputDeviceManager.py", line 79 in _updateThread
  File "/Users/ultimaker/build/env/4.13/install/lib/python3.8/threading.py", line 870 in run
  File "/Users/ultimaker/build/env/4.13/install/lib/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/Users/ultimaker/build/env/4.13/install/lib/python3.8/threading.py", line 890 in _bootstrap

Thread 0x00007000110c7000 (most recent call first):
  File "/Applications/Ultimaker Cura.app/Contents/MacOS/plugins/plugins/RemovableDriveOutputDevice/RemovableDrivePlugin.py", line 61 in _updateThread
  File "/Users/ultimaker/build/env/4.13/install/lib/python3.8/threading.py", line 870 in run
  File "/Users/ultimaker/build/env/4.13/install/lib/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/Users/ultimaker/build/env/4.13/install/lib/python3.8/threading.py", line 890 in _bootstrap

Thread 0x0000700010041000 (most recent call first):
  File "/Users/ultimaker/build/env/4.13/install/lib/python3.8/threading.py", line 306 in wait
  File "/Users/ultimaker/build/env/4.13/install/lib/python3.8/site-packages/zeroconf/__init__.py", line 2534 in wait
  File "/Users/ultimaker/build/env/4.13/install/lib/python3.8/site-packages/zeroconf/__init__.py", line 1715 in run
  File "/Users/ultimaker/build/env/4.13/install/lib/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/Users/ultimaker/build/env/4.13/install/lib/python3.8/threading.py", line 890 in _bootstrap

Thread 0x000070000f03e000 (most recent call first):
  File "/Users/ultimaker/build/env/4.13/install/lib/python3.8/threading.py", line 306 in wait
  File "/Users/ultimaker/build/env/4.13/install/lib/python3.8/threading.py", line 558 in wait
  File "/Applications/Ultimaker Cura.app/Contents/MacOS/plugins/plugins/UM3NetworkPrinting/src/Network/ZeroConfClient.py", line 81 in _handleOnServiceChangedRequests
  File "/Users/ultimaker/build/env/4.13/install/lib/python3.8/threading.py", line 870 in run
  File "/Users/ultimaker/build/env/4.13/install/lib/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/Users/ultimaker/build/env/4.13/install/lib/python3.8/threading.py", line 890 in _bootstrap

Thread 0x000070000e03b000 (most recent call first):
  File "/Users/ultimaker/build/env/4.13/install/lib/python3.8/site-packages/zeroconf/__init__.py", line 1369 in run
  File "/Users/ultimaker/build/env/4.13/install/lib/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/Users/ultimaker/build/env/4.13/install/lib/python3.8/threading.py", line 890 in _bootstrap

Thread 0x000070000d038000 (most recent call first):
  File "/Users/ultimaker/build/env/4.13/install/lib/python3.8/threading.py", line 306 in wait
  File "/Users/ultimaker/build/env/4.13/install/lib/python3.8/threading.py", line 558 in wait
  File "/Users/ikkez/Library/Application Support/cura/4.13/plugins/MKSWifiPlugin/MKSWifiPlugin/MKSOutputDevicePlugin.py", line 211 in _handleOnServiceChangedRequests
  File "/Users/ultimaker/build/env/4.13/install/lib/python3.8/threading.py", line 870 in run
  File "/Users/ultimaker/build/env/4.13/install/lib/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/Users/ultimaker/build/env/4.13/install/lib/python3.8/threading.py", line 890 in _bootstrap

Thread 0x000070000c035000 (most recent call first):
  File "/Users/ultimaker/build/Cura/4.13/build/inst/lib/python3.8/site-packages/UM/Backend/Backend.py", line 165 in _storeStderrToLogThread
  File "/Users/ultimaker/build/env/4.13/install/lib/python3.8/threading.py", line 870 in run
  File "/Users/ultimaker/build/env/4.13/install/lib/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/Users/ultimaker/build/env/4.13/install/lib/python3.8/threading.py", line 890 in _bootstrap

Thread 0x000070000b032000 (most recent call first):
  File "/Users/ultimaker/build/Cura/4.13/build/inst/lib/python3.8/site-packages/UM/Backend/Backend.py", line 153 in _storeOutputToLogThread
  File "/Users/ultimaker/build/env/4.13/install/lib/python3.8/threading.py", line 870 in run
  File "/Users/ultimaker/build/env/4.13/install/lib/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/Users/ultimaker/build/env/4.13/install/lib/python3.8/threading.py", line 890 in _bootstrap
[... 59 lines cut ...]
[... cut ...]
  File "/Users/ultimaker/build/Cura/4.13/build/inst/lib/python3.8/site-packages/UM/Qt/QtApplication.py", line 396 in exec_
  File "/Users/ultimaker/build/Cura/4.13/build/inst/lib/python3.8/site-packages/cura/CuraApplication.py", line 877 in run
  File "/Users/ultimaker/build/Cura/4.13/build/inst/bin/cura_app.py", line 236 in <module>
  File "/Users/ultimaker/build/env/4.13/install/lib/python3.8/site-packages/cx_Freeze/initscripts/Console.py", line 36 in run
  File "/Users/ultimaker/build/env/4.13/install/lib/python3.8/site-packages/cx_Freeze/initscripts/__startup__.py", line 74 in run
```

GregValiant, 2026-09-07:
I'm cleaning up these old reports.  I will go ahead and close this.
If the problem persists into Cura 5.13.0 please fill out a new report so it goes to the front of the queue.

closed 2026-09-07 by GregValiant