Ultimaker/Cura #10950: A 2017 document ... A GPU-based Parallel Slicer for 3D Printing

physicsimulation, 2021-11-28:
### Is your feature request related to a problem?

Cura Engine Performance issue

### Describe the solution you'd like

https://www.researchgate.net/publication/322562978_A_GPU-based_parallel_slicer_for_3D_printing

### Describe alternatives you've considered

Formware 3d but exclusive to resin 8(

### Affected users and/or printers

All

### Additional information & file uploads

All modern softwares shall use GPU when available.
I know that Cura interface uses GPU when moving display representation.
Cura Slicer Engine uses a single CPU core and under exploits my 8 core CPU without talking about my available GPU power.

nallath, 2021-11-29:
The slicing (eg; cutting up the object into smaller parts) is only ~3% of the time that is spent by CuraEngine. Passing that to the GPU won't significantly improve the speed.

fieldOfView, 2021-11-29:
To add to that, other tasks CuraEngine performs can be hard to run in parallel. For example creating supports is done from the top down, and what needs to be supported in a layer always depends on the result of the layer above it. So it is always a matter of processing one layer after the other, instead of it being easy to process multiple layers at the same time (either on a CPU or on a GPU).

fvrmr, 2021-12-03:
Closing this issue this is not something we want since it will not really improve and it is not easy to implement.

closed 2021-12-03 by fvrmr