Ultimaker/Cura #10951: Cura slow on Windows 11

rv6a, 2021-11-28:
### Application Version

4.12.1

### Platform

Windows 11

### Printer

Creative Ender 5

### Reproduction steps

Cura is taking 5-10 seconds to respond. I removed Cura, scanned and removed any left over Cura files. Response is better but still takes 4-5 seconds to respond.

### Actual results

Response is better but still takes 4-5 seconds to respond.
IE. It took 10 seconds to save the project file.
Trying to manipulate objects is very hard since there is a delay.

### Expected results

Response times improved but still too slow to use
[cura.log](https://github.com/Ultimaker/Cura/files/7614214/cura.log)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[CE5_wall (2).zip](https://github.com/Ultimaker/Cura/files/7614224/CE5_wall.2.zip)

fieldOfView, 2021-11-29:
Since you don't have an Ultimaker printer, you could try to disable the "UM3 Network Printing" plugin on the "Installed" tab of the Marketplace. For some users, this plugin (or the part of that plugin that discovers Ultimaker printers on the network) seems to cause slowdowns.

rv6a, 2021-11-29:
Thanks for the suggestion. I disabled UM3 Network Printing and nothing has changed.

Reich187, 2021-11-29:
Kira is running extremely slow for me in the preferences tabs. When selecting my profiles for different types of plastics. I either get not responding or it takes usually a minute or two for a option to be changed after clicking on it.

fvrmr, 2021-11-30:
Hi @rv6a thank you for your report.
I need to check this with my colleague who has Win11
Could you check if it is not slow when opening a file from your file explorer? See: https://github.com/Ultimaker/Cura/issues/10936

rv6a, 2021-11-30:
It is the same no matter if I open it from file explorer of from within
Cura.

Ghostkeeper, 2021-12-08:
From the log file we don't really see anything special going on. The operations that it logs the time for are completed quickly (e.g. rotating an object is like 2ms). I'd guess it's the rendering that makes it slow. You could try setting the layer view to use compatibility mode (in the preferences). It not just influences layer view but causes all of Cura to downgrade to OpenGL 2.0 instead of 4.1. Maybe that'd help.

We haven't been able to reproduce this issue yet, but we currently don't support Windows 11 yet, because the GUI platform we're using doesn't support it yet: https://doc.qt.io/qt-5/supported-platforms.html . Not even the most modern version of Qt supports it yet (https://doc.qt.io/qt-6/supported-platforms.html).

rv6a, 2021-12-12:
I changed the layer view to use compatibility mode. That did seem to help a
lot. I did notice that if I click on any of the dropdowns (i.e. Mesh Fixes,
Special Modes) on the right and click on the preferences icon, the system
goes unresponsive again. I can not click on the "X" in the top right-hand
corner to close Preferences. I can close it with the Close button. Then
when I clicked on Slice it took 5+ seconds to respond with  Preview/Save to
Disk to appear.

nallath, 2021-12-13:
We have created a special plugin to profile where time is being spent on. You can find it here:https://marketplace.ultimaker.com/app/cura/plugins/UltimakerPackages/ExtensiveSupportLogging

Could you use this plugin and share the information that it generates with us?

rv6a, 2021-12-13:
I installed the plugin. What folder is the zip file created in? I can not
find it.

AnSch1510, 2021-12-16:
I've probably the same issue. The whole UI is totally unresponsive.
As you mentiond, i've installed the plugin

> We have created a special plugin to profile where time is being spent on. You can find it here:https://marketplace.ultimaker.com/app/cura/plugins/UltimakerPackages/ExtensiveSupportLogging
> 
> Could you use this plugin and share the information that it generates with us?

@rv6a to generate the zip, you will find the interface to interact with the extension the top of the window on "Extensions" -> "Remote Support"

you will find my file attached:
[remote-support_21-12-16-20_57_31.zip](https://github.com/Ultimaker/Cura/files/7730095/remote-support_21-12-16-20_57_31.zip)

nallath, 2021-12-17:
> I've probably the same issue. The whole UI is totally unresponsive.

I see that changing the active tool takes a _very_ long time for you. You did it twice and it took > 1.5s each time that you did it. I'm honestly a bit baffled that this is the bottleneck. Then again, it's good that we now have the information, because I would have never suspected that bit of code to be the culprit! 

A few questions / requests:
1. Could you try this again but force compatibility mode (you can do this in preferences)
2. Could you make another version where you have it in normal mode, but select /deselect the model on the build plate a few times?

rv6a, 2021-12-18:
Hi Jaime,
In the previous instance, Cura was already in "force layer view
compatibility mode". I turned it off (normal mode) and restarted. I then
opened my file, click on it several times and also reoriented it. Please
let me know if you want me to do anything else.
Regards,
Don Mack

AnSch1510, 2021-12-19:
Hi @nallath ,
I've done another test while enabling "compatibility mode".

See my preference panel:
[image]

The recorded remote-support file:
[remote-support_21-12-19-21_18_59.zip](https://github.com/Ultimaker/Cura/files/7741528/remote-support_21-12-19-21_18_59.zip)

and a screen recording: (the dropdown menu is somehow not visible on the recording, but for me)
https://user-images.githubusercontent.com/9435005/146689649-9564735d-c91b-4eaf-b668-64214fade8d8.mp4

I hope this will help
Best Regards

AnSch1510, 2021-12-19:
I just got notified that a windows update is avilable - after installing it, cura runs normal again... So just a Windows11 bug...

rv6a, 2021-12-20:
I had my windows update run this weekend as well. All seems fine now with
Cura.

nallath, 2021-12-20:
Well, I'm glad that it was an issue with Windows, because I couldn't find out what was causing this on our side.

Thanks for all the help trying to resolve this!

rv6a, 2021-12-31:
The problem started happening again. I found that giving the Cura program Administrator Privileges in Windows fixed the problem.

sbsoneji, 2022-01-03:
Problem still exists it is extremely slow, it takes 5-10 seconds to respond and 10-15 seconds just to enter the manage materials settings. Tried running it as admin but nothing fixed, I am on latest version of windows 11

AnSch1510, 2022-01-15:
Damn, on my PC the slow behavior is back, too

AnSch1510, 2022-01-15:
seems to be fixed after the last windows update... this is bug so annoying

simulz, 2022-02-19:
4.12.1
Windows 11 에서 CURA가 심하게 멈춥니다.
다시 활성화 되고 또 멈추고, 무한 반복

bertenvdb, 2022-02-21:
This should be re-opened IMHO as this isn't fixed.I tried every suggested solution above and anything I could find on other forums, but Cura just doesn't work as it should.  

My PC is pretty powerful. Other CPU intensive and memory hungry applications work on my machine without any problems, no mather what Windows update I do. Only Cura is unusable slow. 

Opening Cura, loading a model, slicing and saving can take up to 10 minutes. Oh wait, a freeze. Let's wait 5 minutes. No response? Force close and repeat. Damn, this model needs supports. Add 3-5 minutes just to enable this. Pfff... 

Please reopen this and fix this.

fvrmr, 2022-02-24:
Reopened issue, since it is not solved for everyone. 
Could you install the extensive logging plugin? https://marketplace.ultimaker.com/app/cura/plugins/UltimakerPackages/ExtensiveSupportLogging and share the logs created from the plugin?

Marmol95, 2022-03-03:
Hola a todos yo tambien estoy teniendo problemas  de lentitud con Ultimaker Cura 4.13.1 y windows 11. Creo haber encontrado una posible solucion y es habilitar los permisos de control total para nuestro usuario.
[image]

closed 2022-03-21 by no-response[bot]