Ultimaker/Cura #10953: Crash When Loading Preview

nickrooney, 2021-11-29:
### Application Version

Beta 2 99.10.0

### Platform

Windows 10

### Printer

Ender 3

### Reproduction steps

I add any stl, then slice it. After I slice it, I go to the preview tab, it loads to 99% and then crashes

### Actual results

Crashes once it hits 99%

### Expected results

It crashes

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Happens with any file, and persisted after restarts

fvrmr, 2021-11-29:
Hi @nickrooney thank you for your report.
Could you share your log file? You can find your log file here: `C:\Users\<your username>\AppData\Roaming\cura\<Cura version>\cura.log`

nickrooney, 2021-11-30:
I can only assume that the version is 0.0? I will attach that log file here, let me know if that is not the correct version for this beta.
[cura.log](https://github.com/Ultimaker/Cura/files/7622723/cura.log)

fvrmr, 2021-12-07:
Thank you for your log file. Your log file doesn't give me more information why this is happening unfortunately.
You can slice in 4.12.1?

Ghostkeeper, 2021-12-20:
This issue is known to occur when slicing something so big that the layer data doesn't fit in VRAM.

If this is the issue, it should be recognisable by that it doesn't crash on smaller prints.

Another issue in the beta (since fixed) is that apparently we were reading out of memory due to a bug in the AMD drivers. Are you using AMD?

closed 2021-12-21 by no-response[bot]