Ultimaker/Cura #10954: Cura not recognizing Monoprice Mini v2 Printer

dukenukemx, 2021-11-29:
### Application Version

4.12.1

### Platform

Linux Mint 20

### Printer

Monoprice Maker Select Mini v2

### Reproduction steps

Run Cura and select Monoprice Maker Select Mini v2 (E3D).  Cura says, "In order to monitor your print from Cura, please connect the printer".  Proterface will detect it and works with my printer.  

### Actual results

Nothing happens since it can't see the printer.

### Expected results

That it would detect the printer like it does in Windows and operate the printer.  

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I don't have anything else to add.

fvrmr, 2021-11-29:
Hi @dukenukemx thank you for your report.
We don't maintain USB printing anymore since our printers don't use USB.
You could check if the USB printing plugin is disabled. Go to Marketplace - Installed 
[image]

dukenukemx, 2021-12-01:
Yes USB printing is enabled.

dukenukemx, 2021-12-02:
This is what I see in the log file.  

2021-12-02 07:41:32,428 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [213]: Error check finished, result = False, time = 4.2s
2021-12-02 07:41:33,773 - DEBUG - [JobQueueWorker [1]] USBPrinting.AutoDetectBaudJob.run [49]: Checking /dev/ttyACM0 if baud rate 250000 works. Retry nr: 0. Wait timeout: 3
2021-12-02 07:41:34,242 - DEBUG - [MainThread] UM.Controller.setActiveStage [180]: Setting active stage to MonitorStage
2021-12-02 07:41:38,277 - DEBUG - [JobQueueWorker [1]] USBPrinting.AutoDetectBaudJob.run [49]: Checking /dev/ttyACM0 if baud rate 500000 works. Retry nr: 0. Wait timeout: 3
2021-12-02 07:41:41,324 - DEBUG - [MainThread] cura.AutoSave._onTimeout [64]: Autosaving preferences, instances and profiles took 0.020487308502197266 seconds
2021-12-02 07:41:42,781 - DEBUG - [JobQueueWorker [1]] USBPrinting.AutoDetectBaudJob.run [49]: Checking /dev/ttyACM0 if baud rate 230400 works. Retry nr: 0. Wait timeout: 3
2021-12-02 07:41:47,285 - DEBUG - [JobQueueWorker [1]] USBPrinting.AutoDetectBaudJob.run [49]: Checking /dev/ttyACM0 if baud rate 76800 works. Retry nr: 0. Wait timeout: 3
2021-12-02 07:41:51,791 - DEBUG - [JobQueueWorker [1]] USBPrinting.AutoDetectBaudJob.run [49]: Checking /dev/ttyACM0 if baud rate 57600 works. Retry nr: 0. Wait timeout: 3
2021-12-02 07:41:56,296 - DEBUG - [JobQueueWorker [1]] USBPrinting.AutoDetectBaudJob.run [49]: Checking /dev/ttyACM0 if baud rate 38400 works. Retry nr: 0. Wait timeout: 3
2021-12-02 07:42:00,801 - DEBUG - [JobQueueWorker [1]] USBPrinting.AutoDetectBaudJob.run [49]: Checking /dev/ttyACM0 if baud rate 19200 works. Retry nr: 0. Wait timeout: 3
2021-12-02 07:42:05,307 - DEBUG - [JobQueueWorker [1]] USBPrinting.AutoDetectBaudJob.run [49]: Checking /dev/ttyACM0 if baud rate 9600 works. Retry nr: 0. Wait timeout: 3
2021-12-02 07:42:24,825 - DEBUG - [JobQueueWorker [1]] USBPrinting.AutoDetectBaudJob.run [49]: Checking /dev/ttyACM0 if baud rate 115200 works. Retry nr: 1. Wait timeout: 15

dukenukemx, 2021-12-02:
Fixed the problem.   It's a 3 year old bug you guys haven't gotten around to fixing.  The bug still exist in Cura 4.12.1.  

https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/10

dukenukemx, 2021-12-03:
I should also note that while it can see the printer now it doesn't seem to operate it.  Might be due to the fact that I extracted the AppImage and then modified the AutoDetectBaudJob.py.  So it could be a permissions issue or something else.

Ghostkeeper, 2021-12-08:
Right, so it is a duplicate of https://github.com/Ultimaker/Cura/issues/6154.

These issues are not issues we can really fix. We don't have the printer you're using to test with. The suggested workaround has the potential to break the connection for other printers, too. This sort of issue can only really be fixed by the community around the printer that has the connection issue, and if a solution is to be contributed to Cura, it needs to (within reason) keep the behaviour the same for the printers where the connection works.

closed 2021-12-08 by Ghostkeeper

dukenukemx, 2021-12-08:
> Right, so it is a duplicate of #6154.
>
> These issues are not issues we can really fix. We don't have the printer you're using to test with. The suggested workaround has the potential to break the [...]

Considering this issue effects multiple printers and even effects users on Mac OSX then I assume it isn't limited to just my printer and me using Linux.  The original person who posted the issue is using a  Ender 3 with SKR mini E3 mainboard, which is not the same as my Monoprice Mini V2.  The other person who reported this issue is using a Anet A8 PLUS V1.6.  

The community told me to use Pronterface because Cura isn't for printer, just for slicing.  It works but the UI looks like it was made in the 90's.

Ghostkeeper, 2021-12-13:
> Considering this issue effects multiple printers and even effects users on Mac OSX then I assume it isn't limited to just my printer and me using Linux.

Probably not. But we also see millions of slices being done (or at least started) successfully via USB cable. So it's at least limited to some printers or some computers. And last I checked it still worked for my Ultimaker 2, which was admittedly quite a while ago but not much changed to USB discovery since.