Ultimaker/Cura #10955: Unable to adjust printer settings, settings pane is missing

Dloranger, 2021-11-29:
### Application Version

4.12.1

### Platform

Windows 10 64bit

### Printer

Custom

### Reproduction steps

Create a new printer using the GUI

1) non-networked printer
2) Custom
3) Changed name
4) Add

Now, I have no where to actually set the printer settings (like setting it to have 3 extruders, etc). 

I also have an imported printer with 3 extruders, no ability to tweak any of the settings

### Actual results

[image]

Printer shows as added, but no settings pane is available
[image]

### Expected results

Settings panel for the printer itself should be available for configuring the HW details

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/7616136/cura.log)

no project file required, empty field

fieldOfView, 2021-11-29:
Could you save and upload a project file? Even a project file saved with no model on the buildplate would be helpful.

fieldOfView, 2021-11-29:
You seem to have disabled a lot of plugins. Much of the functionality in Cura is actually a plugin. The panel to set the printer settings for example is provided by the "Machine Settings Action" plugin, which you have disabled. I would reconsider the plugins you have disabled.

Dloranger, 2021-11-29:
[Cura Empty Project.zip](https://github.com/Ultimaker/Cura/files/7619217/Cura.Empty.Project.zip)
Adding requested empty project file

Dloranger, 2021-11-29:
> 

Yes, I have disabled a lot of the plugins that are not needed such as upgrading from 4.x.x to the current, I started with a clean install of 4.10 and just upgraded to the 4.12.1.

I have historically had HORRIBLE performance issues due to a very old laptop (opening cura took over 1 minute alone).  I have since upgraded to a near bleeding edge gaming laptop, but still like to keep things minimalistic as much as possible.  In the installer I am rerunning the process, the only thing I recall deselecting is the arduino drivers, and not selecting to have cura associated with most of the file type, don't use them anyways.  From there I hit install and it takes off on its own

I am not seeing the ability to reset the plug-ins activation choices on the installer to enable the plug-ins.

fieldOfView, 2021-11-29:
You normally re-enable the plugins in the same place where you have disabled them; on the "installed" tab of the Marketplace. Reinstalling Cura will not change your configuration, and the enabled/disabled plugins are part of your configuration.

If you want to start Cura with a fully clean slate, start Cura and go to Help -> Show configuration folder. After closing Cura, remove all the contents of that folder, but leave the empty configuration folder itself in place. Then restart Cura.

fieldOfView, 2021-11-29:
> but still like to keep things minimalistic as much as possible.

By the way, you also had the ability to preview the print disabled, and support for writing gcode to sd-cards. Might I suggest not just willy-nilly disabling parts of Cura that you don't understand the repercussions of?

Try the Startup Optimiser plugin from the Marketplace. It is a much safer way to make Cura start up quicker.

Dloranger, 2021-11-29:
fair enough on the plug-ins, I will try the optimizer, this machine may not care nearly as much as the old one.

I also have reset the profile (kept a back-up zip just in case) and will go from here

closed 2021-11-29 by Dloranger