Ultimaker/Cura #10956: Arachne Beta 2 - "Slicing Failed" Report a bug

GregValiant, 2021-11-29:
### Application Version

Arachne Beta 2

### Platform

Windows 10 Home (onboard video OpenGL 4.0 in compatibility mode)

### Printer

Creality E3Pro

### Reproduction steps

Open an STL
Click on Slice

### Actual results

Receive the popup "Slicing Failed" report a bug
Reported a bug

### Expected results

The block would slice
[cura.log](https://github.com/Ultimaker/Cura/files/7617764/cura.log)
[Crash on Slice X-Y Ref Block.zip](https://github.com/Ultimaker/Cura/files/7617774/Crash.on.Slice.X-Y.Ref.Block.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Beta 1 ran well.  I am not having problems with 4.12.1 or previous versions.
A clean re-install did not help.

nallath, 2021-11-29:
Err. That's weird. It's complaining that it can't find a setting that was introduced in Arachne. It's almost as if you're mixing the old frontend up with the new engine...

GregValiant, 2021-11-29:
I've never had an issue with any other update.  I un-installed the beta 1 version and then installed the new version.  After the first install didn't work I went back and deleted the folder from the "roaming" directory and did what I thought was a clean re-install.  Same problem.  It doesn't matter which printer I choose, slicing fails.
I have a few previous Cura versions installed.  Could there be a conflict?

Any suggestions?

fvrmr, 2021-11-30:
> I have a few previous Cura versions installed. Could there be a conflict?

Hmm no that shouldn't be a problem. 
I could slice your project file with no problems. 

Just to double check, in your program folder the beta 1 folder is also deleted right?

nallath, 2021-11-30:
It is using the right Cura Engine, that much is clear from the logs:  "Cura_SteamEngine version 99.10.0-Arachne_engine_beta_2"

GregValiant, 2021-11-30:
@fvrmr and @nallath 
I'm going to do another re-install.  I have several definition files I customized and after I install a new version of Cura I copy them into the new Cura folders.  There hasn't been a problem with this before and I haven't changed anything since 4.10.  I'm going to go through them because I think (at least) one is being the problem.

After the re-install and without touching anything...
No desktop shortcut was created.  When I manually create one the icon is wrong.  The icon shown in the Start strip when Cura is running is also wrong.
When I start Arachne the popup shows up asking me to reset all my printers and start over.  I did that and Arachne runs and slices.  My custom materials and profiles are gone.

Here is another log file for you to peruse.  I know neither of you has anything better to do than help a doddering old man with what is starting to look like an operator error.  The effort is appreciated.  Going through the custom files line-by-line is going to take a while.  If you have any thoughts please pass them along.
Thanks,
Greg
[cura.log](https://github.com/Ultimaker/Cura/files/7626931/cura.log)

GregValiant, 2021-12-01:
This is the sort of thing that happens when an amateur hacks away at software.  One of the files I was copying in from a previous version was an older fdmprinter definition file and Arachne couldn't find all the settings it was looking for so I'm sure some were illegal.  So @nallath that was an excellent call you made.

As a penance for bothering you and @fvrmr - I will deprive myself of my second bowl of ice cream tomorrow.

closed 2021-12-01 by GregValiant

GregValiant, 2021-12-01:
It took a minute but I re-figured out why I was using an older fdmprinter file.
I had changed the label of machine_max_feedrate_e because "Maximum Feedrate" offended my sense of balance.  Then you guys went and made an official change so what I was doing was totally unnecessary.  When you're dumb, ya suffer.