Ultimaker/Cura #10957: Arachne Beta (1 & 2) Concentric infill pattern generation: messy line order

neutebach, 2021-11-29:
### Application Version

Arachne beta (1 and 2)

### Platform

Windows 10

### Printer

CFFFP (E3D Toolchanger)

### Reproduction steps

In Cura Arachne beta, use concentric infill where it can't be avoided (prime tower) or is beneficial for surface quality (support interface). 

### Actual results

When using concentric infill pattern, the order of the lines is often messy, e.g. it doesn't follow a strict order (inside-out or outside-in), but jumps around back and forth. Also, the starting point "doesn't make sense", e.g. it doesn't jump to the closest possible point when finishing a line, but travels around to start a new line, sometimes just a few millimeters, sometimes more, adding unneccessary travel movements and retractions.

This applies everywhere where concentric infill pattern is applied, so for me mostly in support interfaces (where it leads to artefacts in the part surface) and the bottom of the prime tower (where it in the worst case can affect the build plate adhesion of the prime tower). 

prime tower first layer:

[image]

random layer (see layer 63 for reference --  I chose concentric to highlight the issue, wouldn't usually use it for anything aside support interface)

[image]

it may have gotten worse or more random in beta 2 over beta 1, but I can't give an example ATM.

### Expected results

compare same project file loaded in Cura 4.11:

[image]

--> infill lines are processed one after another without jumping back and forth

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

project file attached to reference the mentioned layer
[test-project.zip](https://github.com/Ultimaker/Cura/files/7618502/test-project.zip)
s

fvrmr, 2021-11-30:
Hi @neutebach thank you for your report.
I will bring this up for discussion with the team. Keep you posted!

fvrmr, 2021-12-07:
I have discussed this with the team and created a ticket on our backlog for this.
Devs see CURA-8738

BarsMonster, 2022-12-30:
I confirm that this reproduces in 5.2.1.
I wanted to do Z offset finetuning, but was completely puzzled by unexpected random order of the concentric lines :-)