Ultimaker/Cura #10960: Print One at a Time bug, Ender 3 Pro, Cura 4.12

Boomcoach, 2021-11-29:
### Application Version

4.12

### Platform

Windows 10

### Printer

Creality Ender 3 Pro

### Reproduction steps

I placed two small models on the build plate, selected the One-at a Time" option and Cura gave a notice that it reduced the height of the build volume to allow the gantry room, but it reduced to to about 4 CM height, lower than the models themselves and would not slice.

I have been doing this is version 4.11 with no issues.

### Actual results

Error slicing

### Expected results

Should have sliced the models.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

None

TwomD, 2021-11-29:
check what you have set as Gantry Height in the Machine settings under Extruder 1.
because that's how high Cura sets the build volume.

Ghostkeeper, 2021-12-08:
The reason why the build volume is reduced in height is because after printing the first object, the gantry of the print head can hit the previously printed object while printing the one after.

It's not advisable to change the gantry height unless you know what you're doing.

This seems to be working as expected.

closed 2021-12-08 by Ghostkeeper