Ultimaker/Cura #10963: First layer feedrate is not set when firmware retraction is used

gemakochi, 2021-11-30:
### Application Version

4.11.0

### Platform

Linux

### Printer

Custom FFF printer

### Reproduction steps

1. install printer settings plugin
2. enable firmware retraction
2. slice

### Actual results

generated gcode does not have F parameter in G1 code. print starts with previous F parameter set in file, if used in start codes. this happens only in first layer, other layers have F parameter set correctly. 

```
...
;LAYER:0
M107
;MESH:RetractTest
G0 F3600 X119.66 Y112.352 Z0.3
;TYPE:WALL-OUTER
G11
G1 X100.171 Y112.351 E1.02093
G1 X100.171 Y107.648 E1.26729
G1 X119.86 Y107.648 E2.29869
...
;LAYER:1
M106 S255
;TYPE:WALL-OUTER
;MESH:RetractTest
G11
G1 F4200 X119.947 Y107.548 E14.28353
G1 X119.96 Y107.566 E14.28431
G1 X119.96 Y112.434 E14.45432
...
```

### Expected results

F parameter in first print move.

...
;LAYER:0
M107
;MESH:RetractTest
G0 F3600 X119.66 Y112.352 Z0.3
;TYPE:WALL-OUTER
G11
G1 F4200 X100.171 Y112.351 E1.02093
G1 X100.171 Y107.648 E1.26729
G1 X119.86 Y107.648 E2.29869
...

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Project and gcode in zip file
[r9.zip](https://github.com/Ultimaker/Cura/files/7625239/r9.zip)

GregValiant, 2021-12-02:
It's OK with my Ender 3 Pro the active printer, but it is missing when your "A10M Custom" is the active printer.

gemakochi, 2021-12-03:
I was pretty sure changing print speed was causing the issue. After i remove configurations and start from scratch it resolved.

closed 2021-12-03 by gemakochi