Ultimaker/Cura #10965: Can't export more than 1 material

bjoernbusch, 2021-11-30:
### Application Version

4.12.1

### Platform

Mac

### Printer

Creality Ender 3 V2

### Reproduction steps

1. Open Cura
2. Go to Material Settings
3. Select 1 Material and press export
4. Specify filename and click ok - Material gets properly exported
5. Select the next Material and press export
6. Nothing happens.
7. Restart Cura
8. Export works again for one Material.

### Actual results

Nothing happens.

### Expected results

Material should be exported.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/7625321/cura.log)

Same issue as described in #10280

fvrmr, 2021-12-02:
Hi @bjoernbusch thank you for your report. 
I can't reproduce your issue on Windows. What MacOS version are you using?

bjoernbusch, 2021-12-02:
MacOS Monterey 12.0.1

fvrmr, 2021-12-02:
Thank you, I checked on Monterey and I can reproduce. I will bring this up with the team. Keep you posted! 

I will close this issue since it is a duplicate of indeed #10280 and will post the update there so we keep everything in one issue.

closed 2021-12-02 by fvrmr

bjoernbusch, 2021-12-02:
Perfect, thanks, I will follow the other issue.