Ultimaker/Cura #10966: cant open stl from folder

A-Pearce-HLM, 2021-11-30:
### Application Version

4.12.1

### Platform

windows 10

### Printer

Ultimaker S5

### Reproduction steps

Opened Cura
Select the open from disk icon 
select folder on desktop

### Actual results

application crashes

### Expected results

should open the folder??
[cura.log](https://github.com/Ultimaker/Cura/files/7625959/cura.log)
?

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/7625962/cura.log)

fieldOfView, 2021-11-30:
Have you seen https://support.ultimaker.com/hc/en-us/articles/4402996140946-Ultimaker-Cura-crashes-when-opening-any-file ?

A-Pearce-HLM, 2021-12-01:
I have tried the reg and this corrected the error. I do find it hard to believe you would actually ask a client to uninstall another program to fix yours considering it never had the issue beforehand.

Anyway, the error has been resolved. Thank you for the info

Med venlig hilsen / Kind regards

Ashley Pearce

3D Scanning og VR

***@***.***<mailto:>

+45 3169 0682

***@***.***<https://www.hlm.as/>

+45 3212 5115 | www.hlm.as<https://www.hlm.as>
Find os ***@***.***,12.302122,18z/data=!3m1!4b1!4m5!3m4!1s0x46525158324447c1:0x75f5a646f897f854!8m2!3d55.658616!4d12.303219>

nallath, 2021-12-01:
We're not asking people to uninstall another program. We're asking them to disable a hook that another software leaves in the OS that crashes other applications. Cura isn't the only SW affected by this issue.

closed 2021-12-01 by nallath

A-Pearce-HLM, 2021-12-01:
Then I would suggest explaining that in the link sent as it implies you should uninstall it and leave it uninstalled.

Med venlig hilsen / Kind regards

Ashley Pearce

3D Scanning og VR

***@***.***<mailto:>

+45 3169 0682

***@***.***<https://www.hlm.as/>

+45 3212 5115 | www.hlm.as<https://www.hlm.as>
Find os ***@***.***,12.302122,18z/data=!3m1!4b1!4m5!3m4!1s0x46525158324447c1:0x75f5a646f897f854!8m2!3d55.658616!4d12.303219>

nallath, 2021-12-01:
From the page: "There are two options to prevent Ultimaker Cura from crashing:" 

The first is uninstalling it. The second is disabling the feature that Chitubox added to the operating system that causes crashes for other applications.