Ultimaker/Cura #10970: Top/bottom layer flow should be seperated

Ro3dbubs, 2021-11-30:
### Is your feature request related to a problem?

Hello. 
I am strugling with problem that when i print models what is "print in place" model, if there is small gap between models bodies on first layers. Thoose first layers stick together. I found that when flow rate for top bottom/layers is reduced about on 90% then the firs layer is almoust every places seperated. But there comes another issue. Top layer isnt solid anymore. If i could reduce only bottom layer flow i could eleminate elefanth pfoot without making bad top layers. 

I hope that, i write understandable.  ;D
With best regards 
Rolands
Latvia 

### Describe the solution you'd like

Elephant footh

### Describe alternatives you've considered

Didnt find any alternatives.

### Affected users and/or printers

Thoose who struggles with elephant footh.

Asterchades, 2021-12-01:
There's already a separate setting for Initial Layer Flow. Adjusting this will only influence the initial layer (as its name suggests), allowing you to leave known good values in all other places while contending with elephant's foot issues. It's applied as a multiplier on top of the other values - in other words, the default 100% value means it uses your other specified values exactly as they are, even if they're not explicitly set to 100% themselves.

Adjusting the bottom layer flow separately would result in issues with supported areas becoming separated in the same way that adjusting the combined setting currently affects your top layer results.

Ghostkeeper, 2021-12-13:
The Initial Layer Flow and Initial Layer Line Width settings are indeed intended to allow tuning this. Let us know if anything more is needed (and why). But I think those settings allow sufficient control for what you need.

closed 2021-12-13 by Ghostkeeper

Joan1211, 2022-03-31:
Hello. I will continue on this topic regarding the same request. 

Elephant foot is one issue that cannot be solved with only first layer flow.
The problem of elephant foot is comming from the backlash + squizing the first layer and therefore the gantry that bent. 

The first layer is squizzed and then when going up the the next layer (0,2mm) it actually increase of (i.e) 0,1mm. The reason is that the previous layer was squizzed and the gantry was slighly bent. the other reason is also the backlash of Z axis.

The third layer is going 0,15 mm instead of 0,2mm. And we need to wait the forth (or more) layer for the Z axis to be accurate of 0,2mm and stopping creating elephant foot.

For this reason, I think it's good to have settings that allows the user to configure:
- Bottom layers flow;
- Increment of the flow of bottom layers until the X layers.

I would also second the original idea of the post requesting separate flow for top and bottom.
With 3 layers on top/bottom and infill 20%. When I print let's say 50 layers for a cube, the layer 46 is infill. Then there is the first top layers. In this layer, the material is not properly put, because there is lot of empty below (due to the infill 20%).
the second top layer is a bit better. The last layer also better. However if we want a perfect top layer, we need to increase the flow of the last layer to correct the fact that there is some material missing.  Currently the solution is to increase the number of top layers, (4 or 5 or more). But it also increase the number of bottom layers. The second sollution iw to use the "Gradual infill steps" settings.

But it would be so much more convenient to separate top/bottom layers for the flow settings (and for other settings).

Thanks for reading my first comment on Cura Github :)

Looki2000, 2023-06-05:
I have that problem too. I think adding this option should be reconsidered.