Ultimaker/Cura #10974: Cura Crashes with new mac security update

Yadellc, 2021-11-30:
### Application Version

4.12.1 and older

### Platform

Mac OS 10.15.7

### Printer

Ender 3 Pro

### Reproduction steps

when i try to sync cura pr access the market place it crashes
[20211130_113631.mov.zip](https://github.com/Ultimaker/Cura/files/7627887/20211130_113631.mov.zip)
 This happened After macOS Catalina ecurity Update 2021-007 (version 10.15.7)

About my mac:
Mac mini (Late 2012)
Processor: 2.5 GHz Dual-Core Intel Core i5
Memory: 16 GB 1600 MHz DDR3
Graphics: Intel HD Graphics 4000 1536 MB

### Actual results

See attached Video

Below is the bug report that it wants to send to apple:
Process:               cura [13428]
Path:                  /Applications/Ultimaker Cura.app/Contents/MacOS/cura
Identifier:            nl.ultimaker.cura
Version:               4.12 (4.12.1)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           cura [13428]
User ID:               501

Date/Time:             2021-11-30 12:09:22.632 -0600
OS Version:            Mac OS X 10.15.7 (19H1519)
Report Version:        12
Anonymous UUID:        0C047350-5053-6A2F-0DD0-FF99B49C1B80

Sleep/Wake UUID:       16500421-90BB-423C-8731-7DAD61B838A5

Time Awake Since Boot: 110000 seconds
Time Since Wake:       100000 seconds

System Integrity Protection: disabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
abort() called

Application Specific Signatures:
Graphics kernel error: 0xfffffffb

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff72c8a33a __pthread_kill + 10
1   libsystem_pthread.dylib       	0x00007fff72d4ae60 pthread_kill + 430
2   libsystem_c.dylib             	0x00007fff72ba193e raise + 26
3   libsystem_platform.dylib      	0x00007fff72d3f5fd _sigtramp + 29
4   ???                           	000000000000000000 0 + 0
5   libsystem_c.dylib             	0x00007fff72c11808 abort + 120
6   libGPUSupportMercury.dylib    	0x00007fff59d250a5 gpusGenerateCrashLog.cold.1 + 95
[... 1104 lines cut ...]
USB Device: Logitech Wireless Headset
USB Device: USB Receiver
USB Device: Transcend
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: Hub
USB Device: IR Receiver
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
Thunderbolt Bus: Mac mini, Apple Inc., 14.1

### Expected results

i have to restart Cura, but cannot update market place 

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[20211130_113631.mov.zip](https://github.com/Ultimaker/Cura/files/7627897/20211130_113631.mov.zip)

Hollywoodch, 2021-11-30:
I`m experiencing the exact same problem on my Mac since the last update. Plus everything became so slow, to upload, to slice. Before it would take seconds now minutes at best and sometimes it stalls and I`ll have to reboot and try again..
What can be done?

cesardiez, 2021-12-01:
Desde a última atualização estou tendo o mesmo problema e também na hora de abrir o marketplace.

fvrmr, 2021-12-02:
Hi @cesardiez thank you for your bug report.
I will bring this up with the team. Keep you posted!

evtrados, 2021-12-02:
I just installed this security update on my macbook running Catalina, but I could not reproduce the issue. In my case syncing works fine, just like it used to before the update.
Since @cesardiez mentions also the inability to open the marketplace, could you try the solution posted in the below GH issue?
https://github.com/Ultimaker/Cura/issues/10548#issuecomment-932208618

I'm thinking this might have more to do with the SSL certificates rather than the security update itself.

Yadellc, 2021-12-02:
@evtrados I just installed the cert on my mac and am still having the same issue. i dont think its a trust issue wit cura and the OS, but a security limitation mac is implementing locking down the graphics on older machines. This update also affects Fusion360, causing to not even load https://knowledge.autodesk.com/support/fusion-360/troubleshooting/caas/sfdcarticles/sfdcarticles/Fusion-360-crashes-on-launch-on-Mac-OS-10-15-7.html

evtrados, 2021-12-03:
Ah that might be the case then indeed. My macbook is from late 2015, so I can upgrade to Big Sur or Monterey if I want to. Perhaps Apple has imposed these restrictions on earlier hardware then.
In any case, could you share the logfiles from Cura (`/Users/ultimaker/Library/Application\ Support/cura/<version/cura.log`>) so that we check if we can do anything about it?

cesardiez, 2021-12-05:
> I just installed this security update on my macbook running Catalina, but I could not reproduce the issue. In my case syncing works fine, just like it used to before the update. Since @cesardiez mentions also the inability to open the marketplace, could you try the solution posted in the below GH issue? [#10548 (comment)](https://github.com/Ultimaker/Cura/issues/10548#issuecomment-932208618)
> 
> I'm thinking this might have more to do with the SSL certificates rather than the security update itself.

Updated as mentioned but not successful. The problem persists.
Here is the LOG (/Users/$USER/Library/Application\ Support/cura/4.12/cura.log):

2021-12-05 13:37:49,729 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin FileLogger
2021-12-05 13:37:49,743 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin LegacyProfileReader
2021-12-05 13:37:49,758 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin AMFReader
2021-12-05 13:37:49,775 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin SimpleView
2021-12-05 13:37:49,789 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin SolidView
2021-12-05 13:37:49,805 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin FirmwareUpdateChecker
2021-12-05 13:37:49,817 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin SentryLogger
2021-12-05 13:37:49,830 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin SupportEraser
2021-12-05 13:37:49,845 - INFO - [MainThread] UM.Logger.info [128]: Loaded plugin PrepareStage
2021-12-05 13:37:49,853 - DEBUG - [MainThread] UM.PluginRegistry.loadPlugins [396]: Loading all plugins took 2.0128848552703857 seconds
2021-12-05 13:37:49,866 - INFO - [MainThread] UM.VersionUpgradeManager.upgrade [163]: Looking for old configuration files to upgrade.
2021-12-05 13:37:49,939 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [334]: Failed to get file version: ../eSUN_PLA_PRO_Black+%233.xml.fdm_material, skip it
2021-12-05 13:37:49,943 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [334]: Failed to get file version: ../eSUN_PLA_PRO_Black+%232.xml.fdm_material, skip it
2021-12-05 13:37:49,953 - INFO - [MainThread] UM.VersionUpgradeManager.upgrade [177]: Checking and performing updates took 0.08752822875976562
2021-12-05 13:37:50,032 - INFO - [MainThread] UM.Logger.info [128]: Completed loading preferences.
2021-12-05 13:37:50,052 - INFO - [MainThread] UM.Logger.info [128]: Created system tray icon.
2021-12-05 13:37:52,453 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.loadAllMetadata [522]: Loading metadata into container registry took 2.3859682083129883 seconds
2021-12-05 13:37:52,478 - DEBUG - [MainThread] cura.Scene.CuraSceneController.setActiveBuildPlate [142]: Select build plate: 0
2021-12-05 13:37:52,485 - INFO - [MainThread] cura.CuraApplication.run [803]: Initializing machine error checker
2021-12-05 13:37:52,698 - INFO - [MainThread] cura.UI.MachineActionManager.addDefaultMachineActions [74]: Default machine actions added for machine definition [creality_ender3pro]
2021-12-05 13:37:53,008 - INFO - [MainThread] cura.Settings.GlobalStack.addExtruder [202]: Extruder[creality_base_extruder_0 #2] added to [Creality Ender-3 Pro] at position [0]
2021-12-05 13:37:53,010 - INFO - [MainThread] FirmwareUpdateChecker.FirmwareUpdateChecker.checkFirmwareVersion [79]: No machine with name Creality Ender-3 Pro in list of firmware to check.
2021-12-05 13:37:54,228 - DEBUG - [MainThread] cura.Machines.ContainerTree.__getitem__ [120]: Adding container tree for creality_ender3pro took 1.208446979522705 seconds.
2021-12-05 13:37:54,236 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [204]: Begin loading of script: ChangeAtZ
2021-12-05 13:37:54,238 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [204]: Begin loading of script: ColorMix
2021-12-05 13:37:54,239 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [204]: Begin loading of script: CreateThumbnail
2021-12-05 13:37:54,241 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [204]: Begin loading of script: DisplayFilenameAndLayerOnLCD
2021-12-05 13:37:54,242 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [204]: Begin loading of script: DisplayProgressOnLCD
2021-12-05 13:37:54,243 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [204]: Begin loading of script: FilamentChange
2021-12-05 13:37:54,244 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [204]: Begin loading of script: InsertAtLayerChange
2021-12-05 13:37:54,245 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [204]: Begin loading of script: PauseAtHeight
2021-12-05 13:37:54,246 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [204]: Begin loading of script: RetractContinue
2021-12-05 13:37:54,247 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [204]: Begin loading of script: SearchAndReplace
2021-12-05 13:37:54,248 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [204]: Begin loading of script: Stretch
2021-12-05 13:37:54,249 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [204]: Begin loading of script: TimeLapse
[... cut ...]
2021-12-05 13:42:13,811 - DEBUG - [JobQueueWorker [1]] USBPrinting.AutoDetectBaudJob.run [49]: Checking /dev/cu.usbserial-14210 if baud rate 38400 works. Retry nr: 0. Wait timeout: 3
2021-12-05 13:42:18,318 - DEBUG - [JobQueueWorker [1]] USBPrinting.AutoDetectBaudJob.run [49]: Checking /dev/cu.usbserial-14210 if baud rate 19200 works. Retry nr: 0. Wait timeout: 3
2021-12-05 13:42:22,829 - DEBUG - [JobQueueWorker [1]] USBPrinting.AutoDetectBaudJob.run [49]: Checking /dev/cu.usbserial-14210 if baud rate 9600 works. Retry nr: 0. Wait timeout: 3

Abacab7711, 2021-12-08:
I tried changing name of Cura folder to "Old"  no good. Still crashing, I run Windows via VMware fusion and Cura works fine. It's just a pain going back and forth. Anyone have ideas I'm willing to try!

Yadellc, 2021-12-09:
[cura.log](https://github.com/Ultimaker/Cura/files/7685847/cura.log)
Here is my log file. I have tried the renaming of my 4.12 folder and that didn't work either. I even reinstalled 4.11 as that was the last version that worked for me and it still crashes.

Ghostkeeper, 2021-12-13:
Thanks for the logs, both. I'm not really seeing anything special in those logs. Only in Cesardiez' log it was trying to connect to a USB device but failed, but that's not in Yadellc's log so that is probably not the cause.

I think Yadellc's explanation is pretty likely. If so, the new [RHI](https://www.qt.io/blog/graphics-in-qt-6.0-qrhi-qt-quick-qt-quick-3d) of Qt 6.0 may provide a solution. However that would require a complete rewrite of how we render 3D in Cura.

Yadellc, 2021-12-28:
Hey All,

Just wanted to update you on the issue. I just ran another iOS update on my machine and the issue has resolved itself. I am not able to access the market place and update my libraries.