Ultimaker/Cura #10976: Cura Appimage not saving cura.cfg

Blue-Beaker, 2021-12-01:
### Application Version

4.12.0

### Platform

KUbuntu 20.04

### Printer

Creality Ender 3

### Reproduction steps

1.Change some settings related to ~/.local/share/cura/4.12/cura.cfg, such as settings visualibility
2.Close cura and relaunch it.

### Actual results

The file ~/.local/share/cura/4.12/cura.cfg remains unchanged, so does the related settings.

The log file says the configuration is saved,but it doesn't.

### Expected results

The ~/.local/share/cura/4.12.0/cura.cfg is changed, and settings are saved

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/7631208/cura.log)

smartavionics, 2021-12-01:
Shouldn't the path be ~/.config/cura/4.12.0/cura.cfg ?

nallath, 2021-12-01:
> Shouldn't the path be ~/.config/cura/4.12.0/cura.cfg ?

Yep. That's the one it reads. I've just double checked this on my machine.

Blue-Beaker, 2021-12-01:
> Shouldn't the path be ~/.config/cura/4.12.0/cura.cfg ?

On my device it's ~/.local/share/cura/4.12/cura.cfg.

smartavionics, 2021-12-01:
AFAIK, that's an old name and the presence of that file stops it reading the other one?

Blue-Beaker, 2021-12-01:
> 

I tried to rename 4.12 to 4.12.0 , then cura ignores the config files and appears as a fresh install

nallath, 2021-12-02:
But is there a file in ~/.config/cura/4.12/cura.cfg? What happens if you change that?

Blue-Beaker, 2022-01-30:
> But is there a file in ~/.config/cura/4.12/cura.cfg? What happens if you change that?

When I change the file to modify some settings, cura settings are changed.
But when I change settings in cura, this file isn't changed.

Ghostkeeper, 2022-02-14:
Could it be that the file has somehow gotten the wrong permissions?

When Cura is ran as root, the files it creates are also owned by root. Running Cura later as a normal user would prevent modifying the file.

What is the output of running `ls -l ~/.config/cura/4.12/cura.cfg` in a terminal?

Blue-Beaker, 2023-05-05:
I noticed that cura loads config from `~/.local/share/cura/5.2/cura.cfg`, but after config changes it saves to `~/.config/cura/5.2/cura.cfg`.  
When I copy the second file to replace the first file, cura loads modified configs correctly on next boot. And I've found a workaround for this: symlink `~/.config/cura/5.2/cura.cfg` to `~/.local/share/cura/5.2/cura.cfg`. After updating cura, recreate the link for the new version.  
I guess it's a bug about config paths. The fix should be really simple, just save config to the path where it was read from.

IH303, 2024-05-04:
> I noticed that cura loads config from `~/.local/share/cura/5.2/cura.cfg`, but after config changes it saves to `~/.config/cura/5.2/cura.cfg`. When I copy the second file to replace the first file, [...]

I had the same issue. Creating a symlink fixed it for me. Thank you!
I hope the devs will fix this