Ultimaker/Cura #10981: lack of printing time stamp

SidneiCandrade, 2021-12-01:
### Application Version

4.12

### Platform

mac catalina 10.15.7

### Printer

tarantula pro

### Reproduction steps

after slicing, the printing time is no longer displayed,

### Actual results

I would like the printing time to be displayed again

### Expected results

I would like the printing time to be displayed again

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

fvrmr, 2021-12-02:
Hi @SidneiCandrade thank you for your report.
Could you share a project file (file - save project) and share it here on Github in a .zip

SidneiCandrade, 2021-12-02:
> Responder para: Ultimaker/Cura ***@***.***>
> Data: quinta-feira, 2 de dezembro de 2021 06:23

hello, I can, no problem, it turns out that it's not a file, it's all the files that I put to slice that the printing time no longer appears. 

 

 

Sidnei

SidneiCandrade, 2021-12-03:
hello, here are the 3 files which i put in the curative for printing and all without the time data.

SidneiCandrade, 2021-12-03:
I can't attach files

GregValiant, 2021-12-04:
Zip them together and post the zip file.

closed 2021-12-21 by no-response[bot]