Ultimaker/Cura #10985: Cura 4.12.1 MacOSX 12.0.1 Cannot select Model with mouse

tsarak, 2021-12-02:
### Application Version

4.12.1

### Platform

MacOSX 12.0.1

### Printer

Creality Ender 5 Plus

### Reproduction steps

I opened a random STL file from my local drive and tried to select the object either in Prepare or Preview tab.

### Actual results

No selection made. 
Instead I can select ALL objects by right-clicking and selecting Select All Models. BTW in right-click menu only the middle section options are available, ie those that refer to All models.
By selecting 'Select All Models' the model is selected but trying to rotate, mirror, etc nothing happens. Instead the selection of model is cancelled.

### Expected results

Only the obvious! Move, rotate, mirror et.al. the model

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/7640191/cura.log)

fieldOfView, 2021-12-02:
Do you have multiple monitors, with different "retina" scaling factors?

fieldOfView, 2021-12-02:
If you have multiple monitors, does it matter what monitor Cura is on? One thing you can try is to uncheck the option "Restore window position on start" on the General pane of the Cura preferences.

tsarak, 2021-12-02:
You’re the best :-)
Moving the app on the main screen I can do what ever I like ;-)

BTW is there any solution in case I like to use the app in the external display (throught HDMI on a Macbook Pro (mid 2015)?

Best regards
Thanassis

fieldOfView, 2021-12-02:
What you could try is making the external display the "primary" display, and make sure to have that "Restore window position on start" checkbox in the Cura preferences unchecked.

tsarak, 2021-12-02:
Finally the only viable solution is to uncheck the “Restore window position on start’ checkbox.

Thanks a lot for your prompt support.
Cheers

fvrmr, 2021-12-07:
Closing this issue since it is solved!

closed 2021-12-07 by fvrmr