Ultimaker/Cura #10987: Gcode from 3mf different on Linux vs Mac

Wikunia, 2021-12-02:
### Application Version

4.11.0

### Platform

Linux and M1 Mac

### Printer

Custom FFF Printer

### Reproduction steps

1. Open emoji.3mf as project on Linux which was also created on Linux (see https://github.com/Ultimaker/Cura/issues/10986)
2. Slice
3. Save as 3mf_sliced.gcode
4. Open same emoji.3mf as project on M1 Mac (also Cura 4.11.0)
5. Slice
6. Save as 3mf_sliced_mac_m1.gcode

### Actual results

Completely different gcodes i.e different starting position for slicing by ~0.03mm in both directions.
Which unfortunately makes it already impossible to compare gcodes entirely.

### Expected results

Same gcode on both machines due to the same 3mf file.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

The two gcodes and the 3mf file + the log for linux and mac separated (the log for linux also entails the log used for #10986 (so slicing by manually adding the settings)
[cura_bug_report_10987.zip](https://github.com/Ultimaker/Cura/files/7640639/cura_bug_report_10987.zip)

fvrmr, 2021-12-07:
Hi @Wikunia thank you for your report. 
I have checked this with a developer and it is possible that the engine will use different cores when slicing the models. And comparing the g-code files it is indeed 0.03mm which is probably not even noticeable with the printer resolution.  
I will close this issue since this is not something we can fix. I hope you understand.

closed 2021-12-07 by fvrmr

Wikunia, 2021-12-07:
Can you clarify what you mean by different cores? 
It seems that Linux always provides the same results also independent of how many cores are used by the CuraEngine. 
In general I assumed that Cura should produce reproducible results as testing because a nightmare otherwise.

Ghostkeeper, 2021-12-14:
The g-code should indeed be independent of how many cores are used.

I also don't really know what was meant with "different cores". Ultimaker uses "print cores" as their term for nozzle+hot zone, but that doesn't make sense outside of a context of an Ultimaker printer.
It is indeed expected that the g-code output is identical irrespective of how many CPU cores the user has. Otherwise the output is different when sliced on different computers.

The 0.04mm difference is above what we normally consider rounding errors (CuraEngine assumes that rounding errors up to 10um can occur due to magnification effects on rounding errors).

It could be that this is something similar to https://github.com/Ultimaker/Cura/issues/10369, where an edge case seems to be handled differently on different platforms. We use different compilers on each platform which could be the cause here. On Linux we use GCC. On MacOS we use Apple-Clang. On Windows we use MinGW (for CuraEngine) and MSVC (for Savitar and the rest). These compilers may handle edge cases differently due to us unknowingly hitting undefined behaviour. Or perhaps due to compiler bugs even.

I'd be surprised if this is actually a rounding error. I'd expect it's more likely that the two installations have different preferences (which are not in the project file) such as dropping model on build plate or adjusting the size slightly to make it fit in the build volume.

When I try slicing this on Linux with Cura 4.11, I get the same result as you got on Linux. But I'm unable to test on MacOS.

Wikunia, 2021-12-14:
We verified that we have the same settings visually but if there is a way to save settings I would be happy to provide you with that information. Do you mind to reopen this issue as it seems like it is indeed unexpected behavior?

Ghostkeeper, 2021-12-22:
Well, the way to save settings would be to save the project file... Which you did.

thrmttnw, 2022-10-06:
For information, it is the same bteween Mac and PC with Cura 4.8.0, when a 3mf generated on Mac is opened on PC, the two gcode files (Mac vs PC) leads to different printing.

Apart from already mentioned, here and example of the same gcode linenumbers on Mac and PC. 

Looking at the line before TIME_ELAPSED, this element is printed from a starting point a couple of centimeters away from that in the other gcode file.

It means quality will be different, like approved blob placements will not be reproduced accurately.

Can't share files, but what is causing this seems to lead to different slicer decisions, so other models with do this as well.

Mac
snip ---
G1 F1260 X175.146 Y118.303 E1045.31014
G1 X175.234 Y118.215
G0 F7200 X174.89 Y117.375
G1 F900 E1043.31014
G1 F600 Z3.3
;MESH:NONMESH
G0 F7200 X174.892 Y117.374 Z3.3
G0 X39.79 Y154.599
;TIME_ELAPSED:13387.654681
;LAYER:14
;MESH: etc.
G0 X39.79 Y154.599 Z3.5
;TYPE:WALL-INNER
G1 F600 Z3.2
G1 F900 E1045.31014
G1 F1260 X40.419 Y154.365 E1045.32003
G1 X41.035 Y154.103 E1045.32989
snip ---

PC
snip ---
G1 F1260 X175.146 Y118.303 E1045.24126
G1 X175.234 Y118.215
G0 F7200 X174.89 Y117.375
G1 F900 E1043.24126
G1 F600 Z3.3
;MESH:NONMESH
G0 F7200 X174.892 Y117.374 Z3.3
G0 X20.577 Y133.857
;TIME_ELAPSED:13388.408795
;LAYER:14
;MESH: etc.
G0 X20.577 Y133.857 Z3.5
;TYPE:WALL-INNER
G1 F600 Z3.2
G1 F900 E1045.24126
G1 F1260 X20.322 Y134.475 E1045.25111
G1 X20.095 Y135.103 E1045.26095
snip ---