Ultimaker/Cura #10988: Minimum window size on Linux

MarkMLl, 2021-12-02:
### Application Version

4.8

### Platform

Raspberry Pi OS and other Linux

### Printer

Anet A6

### Reproduction steps

Tried to run 4.8 on Raspbian "Bullseye" using a 1024x768 monitor.

### Actual results

Window cannot be reduced to less than 1360x816 if the default window manager (AKA desktop environment etc.) is being used. If KDE is installed it can be reduced to 1120x672 which is still only marginally usable on the sort of monitor which is likely to be dedicated to a 3D printer.

### Expected results

On a laptop PC running Debian "Bullseye" with KDE the window can be reduced to 960x576.

I found the same limitation when I was using Raspbian "Buster" with an AppImage.

There appears to be no visual reason why the application window should not be reduced to this size, i.e. the clickable areas near the top of the window do not overlap.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I've worked around it by putting a substantially larger monitor next to the printer, but this should not be necessary for something that is basically supervisory.

fvrmr, 2021-12-03:
Hi @MarkMLl thank you for your report.
Cura is designed this way to have a limit. So you can't make the screen smaller, this will affect for example fonts. 
So this is something we won't change. If you want to monitor or control your printer via USB you could use PrintRun.
I will close this issue. I hope you understand.

closed 2021-12-03 by fvrmr

MarkMLl, 2021-12-03:
This is ridiculous. It works properly with KDE on a PC, it's a Cura problem in that it is in some way misinterpreting the display or window size.

Ghostkeeper, 2021-12-14:
Cura's interface scales with the font size of your system. The font can be adjusted by the user in their OS to make it legible for people with bad eye sight (accessibility) or different screen pixel resolutions (e.g. 4k monitors) and Cura will listen to that. To make the interface fit, every size is adjusted. This includes the minimum size of the window, lest elements not fit on the window any more.

The minimum size of Cura's main window is set to 80x48em. Typically this is about 12 pixels per em if the screen is not specifically high-DPI. We design the UX to be optimal at 12 pixels per em. On my monitor it's 13 pixels. From your story I can read that your Debian PC indeed has a text size of 12 pixels. But your Raspbian seems to be using a text size of 17 pixels which is quite large (or more reasonable 14 under KDE). I imagine it's meant to work well with the tiny screens that Raspberry Pi computers are typically used for.

So a solution would be to lower the font size in your system. Useful if this Raspbian system is indeed dedicated to the 3D printer. If it's not dedicated, I suggest that you look into passing specific environment variables to adjust the font size just for Cura. This sort of thing is what Linux is really quite good at.

MarkMLl, 2021-12-14:
Thanks, noted. I'm working around this for the moment by (hopefully) sourcing a small but high-resolution monitor... inadequate for general use but OK for positioning items and setting material properties etc.

As an experiment, I installed KDE on the RPi and noted that while Cura was visually identical (i.e. there was no apparent character size difference etc.) I was able to reduce the window size to 1120x672 (with KDE on a laptop I could go to 960x576). I can say with a fair degree of confidence that I've never tried fiddling with the font size on either system.

My suspicion is that this isn't a shell variable, but something being exported from the window manager via an X11 API... but I've never really got to grips with those.

I've also raised this with the RPi people, although not formally reported as a bug. If you were able to provide any further pointers regarding what settings Cura (or possibly Uranium) is checking I'd happily pass that on to them, since I think squashing this one is in everybody's interest.

Ghostkeeper, 2021-12-15:
The size is obtained here through QFontMetrics, as the ascent of the font:

https://github.com/Ultimaker/Uranium/blob/66c8e3db276c6891154c101f9e556ee3fe062c2b/UM/Qt/Bindings/Theme.py#L37

MarkMLl, 2021-12-15:
Thanks for that. I'll take a brief look myself and then pass it on to the RPi people since I think that points the finger fairly firmly at their display manager... a journalist/techie who's watching this comments that they've changed it recently.

My style of debugging uses a 64-channel logic analyzer and a scope. I'm not up to speed on Python etc.