Ultimaker/Cura #10992: Outer wall disappears on one layer

Po-temkin, 2021-12-02:
### Application Version

4.12.0

### Platform

Windows 10

### Printer

Ender 3

### Reproduction steps

1. Slice model
2. Find bug at layer 79

### Actual results

The outer wall disappears only on layer 79

### Expected results

The outer wall is fine on layer 79

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

The outer wall disappears only on layer 79 of my model
Tried on version 4.12.1 with the same result

[XinYiBox.zip](https://github.com/Ultimaker/Cura/files/7644920/XinYiBox.zip
[image]
)

fvrmr, 2021-12-03:
Hi @Po-temkin thank you for your report.
Could share your project file (file - save project)? With a project file it is a lot easier to reproduce your issue.

Po-temkin, 2021-12-03:
[CE3_XinYiBox.zip](https://github.com/Ultimaker/Cura/files/7651215/CE3_XinYiBox.zip)

GregValiant, 2021-12-03:
You will need to turn off Hole Horizontal Expansion (set it to "0").
When a horizontal hole is encountered by Hole Horizontal Expansion then the vertical holes aren't holes anymore.  Turn Hole Horizontal Expansion up to 2 and look at the preview.  You will see how the walls narrow and expand according to whether there is a horizontal hole encountered during the particular layer.

Po-temkin, 2021-12-04:
Thanks for the explanation

fvrmr, 2021-12-07:
I will close this issue, since @GregValiant explanation is indeed correct and will be the solution for your issue.

closed 2021-12-07 by fvrmr