Ultimaker/Cura #10994: Allow post-processing scripts to be stored on print profile

pedrolamas, 2021-12-03:
### Is your feature request related to a problem?

I normally add a couple of post-processing scripts to all my projects (ex: to always have thumbnails as part of the gcode generated by Cura).

However, if I create a new project, the current post-processing scripts are cleared...

### Describe the solution you'd like

I would like to be able to have the post-processing scripts as part of the printer definition or (preferably) the print profile.

### Describe alternatives you've considered

Manually adding the post-processing scripts every time I start a new project...

### Affected users and/or printers

All who use post-processing scripts and would like them defaulted on the printer definitions or print profile?

GregValiant, 2021-12-04:
I have two post-processors I always use and they are attached to the printer.  This is from:
C:\Users\Home\AppData\Roaming\cura\4.12\machine_instances\Creality+Ender-3+Pro.global.cfg
[metadata]
setting_version = 19
type = machine
group_id = d731341b-0dfb-48c8-97b3-d553b5f94108
post_processing_scripts = [**SearchAndReplace**]\\\nsearch = G1 F2100 E-5\\\nreplace = G92 E-3.5\\\nis_regex = False\\\n\\\n
	[**ShowProgress**]\\\ndisplay_total_layers = True\\\ndisplay_remaining_time = True\\\nspeed_factor = 1.1\\\n\\\n

What printer do you have and what operating system?

pedrolamas, 2021-12-04:
Thanks @GregValiant, I just checked and you are absolutely correct...

I was under the impression that creating a new project (File -> New) would clear the post-processing scripts, but that does not seem to be the case.

Given that, this issue does not make sense anymore so I am now closing as it.

closed 2021-12-04 by pedrolamas