Ultimaker/Cura #10997: Highly excessive ironing speed

MichaelObermeier, 2021-12-03:
### Application Version

Arachne_engine_beta_2

### Platform

Windows 10

### Printer

Size modified Anycubic i3 Mega/ Happens on others too

### Reproduction steps

1. Import 40mm cube from .3mf
2. Select Normal profile
3. Enable Ironing

### Actual results

The default ironing speed is exceeded 10x resulting in 133 mm/s speed instead of 13.33 mm/s.

[image]

Decreasing flow to 1% multiplies the speed another 10x resulting in 1333 mm/s.

[image]

### Expected results

The set ironing speed should be respected.
Print speed should not exceed reasonable levels.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

It seems like this might be an issue with Arachne not recognizing ironing as a separate action and trying to increase speed for backpressure compensation.

[IronSpeed.zip](https://github.com/Ultimaker/Cura/files/7648138/IronSpeed.zip)

fvrmr, 2021-12-07:
Hi @MichaelObermeier thank you for your bug report.
I created a ticket on our backlog for this.
Developers see CURA-8737

Ghostkeeper, 2021-12-20:
This is considered a blocker for the stable release.

fvrmr, 2022-05-03:
This is fixed for 5.0. The beta of 5.0 is already out, see here: https://github.com/Ultimaker/Cura/releases/tag/5.0-beta
I will close this issue, since it is fixed.

closed 2022-05-03 by fvrmr