Ultimaker/Cura #11000: Hole Horizontal expansion  delete the print area from the files.

Talha909, 2021-12-03:
### Application Version

4.12.0

### Platform

Windows 10

### Printer

CR 10

### Reproduction steps

I turn on the Hole Horizontal expansion. 

### Actual results

Print area disappear
[image]
[image]
 from the print. 

### Expected results

It suppose only effect the hole on thr print. But it delete the printed areas from the some sides. 
[image]
[image]
[CCR10SPRO_New bltouch after update.zip](https://github.com/Ultimaker/Cura/files/7649559/CCR10SPRO_New.bltouch.after.update.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

I have upload the both files. 
[CCR10SPRO_New bltouch after update.zip](https://github.com/Ultimaker/Cura/files/7649567/CCR10SPRO_New.bltouch.after.update.zip)
Can you please check

Talha909, 2021-12-03:
[image]

GregValiant, 2021-12-03:
Cura cannot tell the difference between a round feature and a square feature.  A hole is a hole whether it is comprised of 3 sides, 8 sides, or 3000 very short sides.  There are no circles - just approximations.
So Hole Horizontal Expansion acts on any enclosed internal vertical feature whether it's square, an octagon, a trischilioi-gon or whatever.
If you want to adjust the printed diameter of the three vertical holes on that shelf then it is probably easier to do that in CAD.  You could try using a Support Blocker as a Mesh Modifier.
Hole Horizontal Expansion works in the X and Y and so has no effect on horizontal holes, just vertical ones.
Set the Hole Horizontal Expansion to 5 and look at the preview.

Talha909, 2021-12-08:
[image]
This is the view when I add 5 in horizontal expansion.

Ghostkeeper, 2021-12-15:
This is indeed expected behaviour.

You have some (rectangular) holes between the panels in that lower section. It's making them bigger. The bigger hole erases some part of the structure around it, which is those panels, in this case.

Hole Horizontal Expansion works with limited scope: It will only expand holes in the layers, i.e. vertical holes.

There is no way that I know of that can neatly achieve what you want to achieve in Cura. You'd have to edit the model to be able to make those horizontal holes bigger.

closed 2021-12-15 by Ghostkeeper