Ultimaker/Cura #11002: Infill Settings Visualization on Menu Dropdown

Wolverine1977, 2021-12-03:
### Is your feature request related to a problem?

When selecting an infill option from the dropdown I do not know what each of the patterns looks like etc.

### Describe the solution you'd like

Add a icon showing the pattern on mouse over or next to the menu selection. 

### Describe alternatives you've considered

Looking online for a visual guide to each option

### Affected users and/or printers

All Prints where custom infill is desired

### Additional information & file uploads

Thank you for the Consideration

fvrmr, 2021-12-06:
Hi @Wolverine1977 thank you for your feature request.
You can download the settings guide: https://marketplace.ultimaker.com/app/cura/plugins/Ghostkeeper/SettingsGuide2

If you hover over the infill pattern setting it will show you detailed overview of all the infill patterns:
[image]

Let me know if that would help you!

Wolverine1977, 2021-12-06:
Thank you - I was not aware of this feature 🙂

[Screenshot (313)]<https://user-images.githubusercontent.com/77000924/144830906-1bbe6080-050a-4b11-8427-40a3561c597e.png>

fvrmr, 2021-12-07:
I will close this issue, since it will solve your request!

closed 2021-12-07 by fvrmr