Ultimaker/Cura #11003: Terrible performance in some dropdowns

sneakythr0ws, 2021-12-03:
### Application Version

4.12.1

### Platform

Linux(Ubuntu 21.10 and previous)

### Printer

Flyingbear ghost 5

### Reproduction steps

Open `Material` or `Nozzle Size` choice dropdown. 

### Actual results

Performance in this dropdown is very very bad. 

### Expected results

Same performance in all part of system. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

This drop down works fine:
[image]
but `Material` and `Nozzle Size` have very very bad performance. This dropdowns have different color scheme(white.I can't screenshot it).
In preferences dialog performance is good.
I have Intel 11gen Tiger Lake Iris Xe gpu.

fvrmr, 2021-12-07:
Hi @sneakythr0ws thank you for your report.
Could you share your log file? You can find your log file here: `$HOME/.local/share/cura/<Cura version>/cura.log`

sneakythr0ws, 2021-12-07:
> Hi @sneakythr0ws thank you for your report. Could you share your log file? You can find your log file here: `$HOME/.local/share/cura/<Cura version>/cura.log`

Sure. But I don't see changes when I use dropdown(I used tail -f)
[cura.log](https://github.com/Ultimaker/Cura/files/7668379/cura.log)

fvrmr, 2021-12-07:
Hi @sneakythr0ws I can't see anything happening in that log. Could you do your reproductions steps: open material and or open nozzle size dropdowns. And after you did that close Cura and again share your log file here.

sneakythr0ws, 2021-12-07:
> Hi @sneakythr0ws I can't see anything happening in that log. Could you do your reproductions steps: open material and or open nozzle size dropdowns. And after you did that close Cura and again share your log file here.

As I said I tried `tail -f cura.log` and monitored log when I clicked dropdown. Nothing happened in log. 
I think it may be hardware acceleration problem. Is it possible to switch to software api?

sneakythr0ws, 2021-12-09:
@fvrmr do you need any extra info?

fieldOfView, 2021-12-09:
Recently some users have reported sluggish UI performance in Cura that seems to be caused by a 3rd party component that monitors the network for Ultimaker printers. If you don't have any Ultimaker printers, you could try disabling that module.

Go to the "Installed" tab of the Marketplace, and scroll down to "UM3 Network Printing" and uncheck the checkmark in front of it. Restart Cura and check if the terrible performance is still there.

sneakythr0ws, 2021-12-09:
> Recently some users have reported sluggish UI performance in Cura that seems to be caused by a 3rd party component that monitors the network for Ultimaker printers. If you don't have any Ultimaker [...]

I tried. But unfortunately no effect. Dropdown controls still have bad performance(

I have problem only in two dropdowns. Look at my screenshots. With dropdowns have different color scheme than main controls and panels. May be this dropdowns use another styles, components, libraries ?

sneakythr0ws, 2021-12-13:
This window has terrible performance too. It takes 5 seconds to enter one character 
[image]

sneakythr0ws, 2021-12-15:
> Could you install and use this plugin to generate some profiling information for us: https://marketplace.ultimaker.com/app/cura/plugins/UltimakerPackages/ExtensiveSupportLogging

[log.zip](https://github.com/Ultimaker/Cura/files/7718203/log.zip)

nallath, 2021-12-15:
Hmm. I don't really see anything that would explain a slowdown. The rendering update itself seems to take only 2ms, so that's well within the range that one would expect

Could you also run the plugin from startup? It might provide a bit more information.

sneakythr0ws, 2021-12-15:
> Hmm. I don't really see anything that would explain a slowdown. The rendering update itself seems to take only 2ms, so that's well within the range that one would expect
> 
> Could you also run the plugin from startup? It might provide a bit more information.

I made some extra tests. The problem appears only on external display with fractional scaling
[image]

Ghostkeeper, 2021-12-20:
That's a good hint. Unfortunately on my current computer, my work laptop, using fractional scaling at 125% still causes it to go to 200%. Seems to be a bug in Ubuntu (20.04) there for me.

At home I have a more modern version of Ubuntu (21.10) so maybe I could try it again there.

sneakythr0ws, 2021-12-21:
> That's a good hint. Unfortunately on my current computer, my work laptop, using fractional scaling at 125% still causes it to go to 200%. Seems to be a bug in Ubuntu (20.04) there for me.
> 
> At home I have a more modern version of Ubuntu (21.10) so maybe I could try it again there.
> 
> I suspect that Qt 6.2 will fix this issue by removing use of Controls 1, which seems to be the places where this issue occurs.

FYI 4.13 beta affected too

Ghostkeeper, 2021-12-22:
Indeed at home the 125% fractional scaling works properly. But the performance of Cura feels normal, even in the drop-downs and pop-ups. I'm not able to reproduce this issue yet.

sneakythr0ws, 2021-12-22:
> Indeed at home the 125% fractional scaling works properly. But the performance of Cura feels normal, even in the drop-downs and pop-ups. I'm not able to reproduce this issue yet.
For me problem appears only on **external** 4k monitor. On build-in laptop 4k display even with factional scaling it works properly. Other software works fine.

Ghostkeeper, 2021-12-27:
Yes, I have a desktop computer so every monitor is an external monitor. I also have a second "monitor", a 4k projector, which doesn't show the problem either.

GregValiant, 2024-11-16:
Is this still an issue in current versions of Cura (5.8.0 and up)?  Can this be closed?

sneakythr0ws, 2024-11-16:
> Is this still an issue in current versions of Cura (5.8.0 and up)? Can this be closed?

Sure. It works fine

closed 2024-11-16 by sneakythr0ws