Ultimaker/Cura #11006: Flow Function not transferring into G-Code 4.12.0

srdobie, 2021-12-04:
### Application Version

4.12.0

### Platform

Windows 10

### Printer

Ender 3 Pro

### Reproduction steps

Modified the Flow percentage in Cura and noticed that it wasn't imported into the printer.  I sliced a project with the percentage changed to 94 in Cura and looked at the G-Code in Notepad++.  I noted that there was no M221 function in the file.
[image]
 

### Actual results

The printer didn't have the correct Flow percentage after starting the print, so I manually set at the printer. 

### Expected results

M112 should be in the g-code file when it is set in cura.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Here is the G-Code file.
[CE3PRO_keurig spacer filet v4.zip](https://github.com/Ultimaker/Cura/files/7654236/CE3PRO_keurig.spacer.filet.v4.zip)

fieldOfView, 2021-12-04:
This is as expected; the "Flow" parameter is not added as an M112 command, but instead all extrusions (E-values) in the gcode are multiplied by Flow/100.

fieldOfView, 2021-12-04:
In other words, your confusion is about who does the extrusion multiplication: CuraEngine or the printer firmware. The way it is implemented in Cura is that the E-values are multiplied by the slicer, and the printer is unaware of that.

srdobie, 2021-12-04:
Thanks for the education.  Just trying to understand the printer, and found this.  Sorry to bother you.

fvrmr, 2021-12-06:
Closing this issue, since this is not a bug. Thanks @fieldOfView for the explanation!

closed 2021-12-06 by fvrmr