Ultimaker/Cura #11009: adding same part again or multiply 1x causes an off build area issue

95gtspeedemon, 2021-12-05:
### Application Version

4.11.0

### Platform

win 10

### Printer

cr10 v3

### Reproduction steps

add sword to plate, use tool to lay flat side down (90*). multiply sword 1x, it goes from a 30min print to a 7 day print and showing a line off screen/build area that i cant see the extent of.
![Capture](https://user-images.githubusercontent.com/29670205/144730304-38d1c62e-5878-42dc-8280-c161510033ee.PNG
[sword large.zip](https://github.com/Ultimaker/Cura/files/7655053/sword.large.zip)
)

### Actual results

it becomes a 6 day print and 736g of filament (its a 10g print)

### Expected results

off build area issue to go away

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[sword large.zip](https://github.com/Ultimaker/Cura/files/7655054/sword.large.zip)

95gtspeedemon, 2021-12-05:
its a profile issue (which its fine with one sword and every other print thus far). use the fine profile and the issue stopped
[image]
[image]
[image]
.

95gtspeedemon, 2021-12-05:
my profile, instead of using the lay flat tool, i turned the model 90* manually. then multiplied by 1, ok this time...

fvrmr, 2021-12-06:
Hi @95gtspeedemon thank you for your report.
Could you share a project file (file - save project)? Your 3mf does not contain the Cura settings.

closed 2021-12-20 by no-response[bot]