Ultimaker/Cura #11011: Y min and Y max in printhead settings swapped

Triangulix, 2021-12-05:
### Application Version

4.8.0

### Platform

Windows 10

### Printer

Custom

### Reproduction steps

Follow the online documentation to set the value for Y min and Y max in the printhead settings.

### Actual results

The printhead outline drawn by Cura does not match the outline of the printhead. The distances on the Y axis are swapped.
Only if I swap Y min and Y max the drawn outline matches the actual printhead.

### Expected results

Program behaviour and documentation should match.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

fvrmr, 2021-12-06:
Hi @Triangulix thank you for your report.
Where did you find documentation about the printhead settings?

Because this is how you should measure the values:
X min: horizontal distance between nozzle tip and the left side of the printhead.
Y min: horizontal distance between nozzle tip and the front side of the printhead
X max: horizontal distance between nozzle tip and the right side of the printhead.
Y max: horizontal distance between nozzle tip and the back side of the printhead.

The X values are the distance from the print head's nozzle to the left and right side of the print head. The Y values are the distances from the print head's nozzle to the front and back side of the print head.

Triangulix, 2021-12-06:
Hi @fvrmr 
Please look at my screenshot and compare it with what you wrote.

To show the problem I used distinctively big and small values for the min and max values of both axis.
As you can see the big values are X max and Y max and the small values are X min and Y min.

For the X axis the values are applied as positive offsets to the X axis. X max extends the outline to the right (higher), X min extends ist to the left (lower).
For the Y axis the values are applied negated: Y min extens the outline to the top (higher) and Y max extends the outline to the bottom (lower).

On some places the Y min value is described like this:
"Y min: This is the distance from the tip of the nozzle to the edge of the printhead in the negative Y direction (towards the front of the printer from the nozzle)."
While being less ambiguous than your description it is also false as you can see in my screenshot.

fvrmr, 2021-12-07:
Ah you are right. I checked with a developer and I will change the tool tips for the printhead settings, since that description does not say anything at all!

So this is how it should be measured: 
X min: horizontal distance between nozzle tip and the left side of the printhead.
Y min: horizontal distance between nozzle tip and the back side of the printhead.
X max: horizontal distance between nozzle tip and the right side of the printhead.
Y max: horizontal distance between nozzle tip and the front side of the printhead.

Triangulix, 2021-12-08:
Thanks, this should help a bit with the issue.

commit 4ff44dc9 (2021-12-22): More expansive description for print head polygon setting

This description is shown for the X-min/Y-min/X-max/Y-max settings in the Machine Settings dialogue, in the tooltip. The new description is written to sort of apply to the polygon but also to those settings in the GUI. It needs to be understandable from the GUI in any case. Being technically correct for the polygon is bonus.

Fixes #11011.
closed 2021-12-22 by Ghostkeeper