Ultimaker/Cura #11012: Top layer added where is none

thomaskilian, 2021-12-05:
### Application Version

4.12.1

### Platform

OS X (M1)

### Printer

Custom

### Reproduction steps

I have a simple model which is sort of a cup-formed cover. When rendering Cura adds a top layer where none should be. Turning it to the side it does not add support at all.

[image]

Slice with slight tilt:
[image]

Slices bottom down:
[image]

### Actual results

Single "ghost" top layer added

### Expected results

Well, no "ghost" top...

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[CFFFP_temp.3mf.zip](https://github.com/Ultimaker/Cura/files/7656305/CFFFP_temp.3mf.zip)

thomaskilian, 2021-12-05:
I just tried that with 4.9 and it slices correctly there.

GregValiant, 2021-12-05:
Under "Mesh Fixes" de-select "Remove All Holes".  The inside of your box constitutes a rectangular hole and so it is being removed.

thomaskilian, 2021-12-05:
Will test that in an hour or so. Anyhow, that option should rather be an opt-in than an opt-out. Makes me wonder that nobody else fell in that trap already.

thomaskilian, 2021-12-05:
There is no such option:
[image]

thomaskilian, 2021-12-05:
Noticed only now that this has been closed. The above is no solution.

fvrmr, 2021-12-06:
Hi @thomaskilian thank you for your report.
Like @GregValiant is mentioned you need to disable `Remove All Holes`

In 4.9.1 it is also showing the ghost layer. Because this is set in your project file. 
[image]

thomaskilian, 2021-12-06:
Well, I just don't see that option :-/ And I rendered the stl file from source in the former versions, not from a project file. Before/including V10 it worked and starting from V11 it fails. Here what V11 shows (working on another machine currently):
[image]

GregValiant, 2021-12-06:
Check the setting visibility.  If you hover the mouse over Mesh Fixes an icon will appear on the right.  Click it and see if Remove All Holes appears in the dialog.
An alternative is to select the Setting Visibility control to the right of the Setting Search box and set the visibility to "All".

[image]

This is with it un-checked.

[image]

thomaskilian, 2021-12-06:
Uhm, interesting. Now on this machine I see the option. I was sure to have looked at the complete list yesterday on the other machine and not seeing it. Well, I'll check that when I switch over later the day...

thomaskilian, 2021-12-06:
Well, in any case: I never touched that option (as you can see I was not even aware of it's existance). So it must have been set during migration. Either on purpose (bad idea) or accidentally (not better either). In any case this option needs to be turned off except for people wanting it to be set (for whatever case).

thomaskilian, 2021-12-06:
So I'm at my machine now and now I see the option (no idea what was on yesterday). However, it's not set:
[image]
And toggling it does not seem to tangent the slicer. It won't re-enable Start Slicing. Also it will not appear in the menu.

[image]

And maybe you have not noticed, but if you see the very first picture above you notice that there was no support generated. There's something very strange here.

GregValiant, 2021-12-06:
Here is a 3mf with your printer.  I've altered some settings but left the retraction distance alone.  When you open it in Cura you will see how I've set it up.  
[GV_temp.zip](https://github.com/Ultimaker/Cura/files/7660991/GV_temp.zip)

thomaskilian, 2021-12-06:
Can you tell me how to open that with Cura??? Cmd-O does not recognize the format (only Prusa Slicer would).

GregValiant, 2021-12-06:
FIle | Open

thomaskilian, 2021-12-06:
I did say that this is not possible. Greyed out:
[image]

fvrmr, 2021-12-07:
You can drag the 3mf also onto the build plate.

thomaskilian, 2021-12-07:
Not really:
[image]

thomaskilian, 2021-12-07:
Interestlingly though, I now see the Hole option:

[image]

fvrmr, 2021-12-07:
Strange... Did disabling this setting solve the issue for you?

thomaskilian, 2021-12-07:
Well, now it slices correctly. Don't ask me, but I'd tell you there'se something terribly fowl and you should keep an eye on this.

closed 2021-12-07 by thomaskilian

Ghostkeeper, 2021-12-16:
> Will test that in an hour or so. Anyhow, that option should rather be an opt-in than an opt-out. Makes me wonder that nobody else fell in that trap already.

It is by default disabled. I don't think any profile in Cura's distribution ever enables Remove All Holes by default. Nor do I think they should. It's a mesh fixing setting, so not really relevant to profiles.
It could be that you loaded a profile or project file from someone else at some point that enabled this setting.

> Interestlingly though, I now see the Hole option:
> 
> [image]

This happens if you disable the 3MF reading plug-in. Some people disable that to improve performance, though it really only saves a miniscule amount of time. I'd suggest the Start-up Optimizer plug-in instead which is much more clever with what it disables.

thomaskilian, 2021-12-16:
I disabled quite some plugins which did tell me nothing. That might have been one of them. Interestingly though dropping the file seemed to have made changes. I can't tell for sure under which circumstance the Mesh option did appear. But since it appeared my issue was gone. Definitely I smeel a bug hiding there somewhere.