Ultimaker/Cura #11013: Camera View Switching

ipodsync, 2021-12-05:
### Is your feature request related to a problem?

Switching between  perspective and orthographic when laying out multiple objects.

### Describe the solution you'd like

In the bottom left, you can easily change the camera position.  It would be nice if this area also provided the ability to toggle the camera view between perspective and orthographic.

### Describe alternatives you've considered

I use the menu bar

### Affected users and/or printers

Anyone laying out multiple parts

fieldOfView, 2021-12-05:
The Sidebar GUI plugin adds this. but it also rearranges the whole UI (into something arguably more sensible).

Ghostkeeper, 2021-12-16:
We considered this before, when the feature was added. We decided against adding buttons to the main view for this, because it's not often that users need to switch between these two camera projections. Most people work either with perspective (to view models realistically) or with orthographic (to measure lengths more easily), not both. Not adding the buttons keeps the interface simpler, not looking as much like a cockpit full of switches.

Of course, the Sidebar GUI plug-in adds it and for people that still prefer to have those buttons that is a good solution.

closed 2021-12-16 by Ghostkeeper