Ultimaker/Cura #11014: Cura 4.12.1 not slicing

Chartroozer, 2021-12-05:
### Application Version

4.12.1

### Platform

windows 10 (i7 6 cores/ 32 gig ram / 1070 gpu/ cura installed on nvme ssd)

### Printer

CR 10 v3

### Reproduction steps

1. put a model on plate
2. slice

### Actual results

Cura does not slice
[image]

### Expected results

It should slice like it was before.  Cura hangs and does not slice , left it for hours and it stays. Cura's ram use is at 1.6% and processing power at 0%  so definitely the issue is linked to this. 

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log.1.zip](https://github.com/Ultimaker/Cura/files/7656356/cura.log.1.zip
[CCR10_Dome_01-trenchR.zip](https://github.com/Ultimaker/Cura/files/7656363/CCR10_Dome_01-trenchR.zip)
)

fvrmr, 2021-12-06:
Hi @Chartroozer thank you for your report.
Could you share your project file (file - save project)? Your 3mf doesn't contain the Cura settings.

Chartroozer, 2021-12-06:
I don t have the file anymore , I uninstalled and reinstalled 4.12.0 and it
works so I will wait for 4.12.2 before updating.
The only thing i have is the screenshot i uploaded showing no cpu resources
used and the log.

fvrmr, 2021-12-07:
Than it is hard for us to see why this is happening. I will leave this issue open and wait till somebody else have reproduction steps or a project file to reproduce this issue.

Chartroozer, 2021-12-07:
I understand, sorry i cannot help further.

Ghostkeeper, 2021-12-16:
The only change to the slicing engine in 4.12.1 is something that prevents Lightning Infill from overextruding when multiple parts are involved. If you were not using Lightning Infill then I think nothing should have changed, and what you were seeing was likely a random occurrence or edge case that could as well have happened in 4.12.0.

Chartroozer, 2021-12-16:
yes probably. Not sure I went back to 4.12 and it works fine.