Ultimaker/Cura #11016: Unneccesary dense infill with newer cura versions

doppelgrau, 2021-12-05:
### Application Version

4.11/4.12.1

### Platform

Linux/Ubuntu 20.04

### Printer

Tenlog TL-D3 Pro

### Reproduction steps

Slice a part that has surfaces on different heights. Simple example attached.
[test.zip](https://github.com/Ultimaker/Cura/files/7656570/test.zip)
Profile used also attached.
[profile.zip](https://github.com/Ultimaker/Cura/files/7656581/profile.zip)

### Actual results

At the height of other surfaces the infill is always printed dense, the "infill steps" are ignored for at least one layer. see pictures
[image]
[image]
.

### Expected results

The infill is not affected, like for example in version 4.8
[image]

[image]

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

It's annoying, since for more complex parts it can significantly increases print time and material consumption. Currently held me back upgrading from 4.8 even if I like the newer features.

nallath, 2021-12-06:
You have gradual infill enabled. What you see is the effect of that.

doppelgrau, 2021-12-06:
Why does the behaviour changed since 4.8? (compare the pictures,with 4.8 the areas with no surface keep the spare infill)
And why can I observe, that sometimes on the areas not affected the infill sense steps are skipped, since only one or two layer are printed dense?

fieldOfView, 2021-12-06:
You have likely enabled the gradual infill at some point in Cura 4.9 or newer and forgotten about it. Then when you installed a newer version, that newer version imported the settings from the previous version, including the gradual infill setting value.

doppelgrau, 2021-12-06:
gradual infill is also enabled in the 4.8 version, and there only the area below the surface is printed denser. not one layer dense for the other areas and then continue sparse.

Ghostkeeper, 2021-12-17:
Cura doesn't have any Tenlog profiles, so I can't exactly reproduce your print.

However I can reproduce this problem with Custom FFF Printer too using this model. I think it's actually unrelated to just gradual infill. The density of the infill is correct. However a few layers higher there is a problem with the stitching, resulting in a layer with a vastly different shape. This causes skin to generate and also the denser steps of gradual infill.

[image]

I think this is a duplicate of https://github.com/Ultimaker/Cura/issues/10369 . It seems to have the same sort of case, and the same effect.

closed 2021-12-17 by Ghostkeeper