Ultimaker/Cura #11017: Markings to indicate changes to settings sections

Garryb123, 2021-12-05:
### Is your feature request related to a problem?

If the user makes 1 or 25 changes to settings a star shows up in the Profile name drop-down. The star tells the user there are changes from the profile default but NOT WHERE THEY ARE. If the user collapses all settings headings it's easy to forget which section a change had been made in and there is no way to know which section has a change but to begin expanding each section.

### Describe the solution you'd like

I suggest and request something like bars, dots, or maybe even better, numbers to indicate changes in any section like this image to provide instant, simple information that any particular section contains changes.
 
[image]

### Describe alternatives you've considered

Otherwise the user must constantly open and close the settings heading (or leave each heading open [but then much scrolling is required]).

### Affected users and/or printers

All printers should benefit from this.

fieldOfView, 2021-12-06:
I like the general idea. One thing that deserves some further thinking is how this should handle cases where settings that are not made visible affect the count.

Garryb123, 2021-12-06:
Indeed; that complexity will require some thought.

After reading other FR's I came back to this and realized I left out something important but obvious. I should have explained this dynamic occurs while testing new settings and obviously ceases to be an issue after saving changes to the profile.

fvrmr, 2021-12-07:
Hi @Garryb123 thank you for your feature request.
We have a similar ticket on our backlog already. I will link this to our ticket!
Devs see CURA-8721

Garryb123, 2021-12-07:
Thank you so much @fvrmr. Your news has made my day and I'll look forward to your implementation. Best regards ..

nallath, 2021-12-08:
I really like this idea.

gonzalo-massa, 2022-05-15:
I just came here to add this as a feature request, and found this open issue. I like the idea although I've never thought of showing how many settings were modified (it's irrelevant to me) but, in that case, I think a number is far better than have the user counting bars or dots.

But if there is any complication or thought needed to se how to count the modified settings, I think a first version of this improvement would be just to have an indicator. This could be a dot or an icon (maybe the same icon you show next to the profile name when it has changes, to have a consistent UI) or just highlight some element like the title of the section with a distinct color, or the caret to open the section, etc.