Ultimaker/Cura #11018: unable to slice

moosehaed, 2021-12-05:
### Application Version

Arcane

### Platform

windows 10

### Printer

anycubic predator

### Reproduction steps

Bug report.

Downloaded and installed Arcane Beta.
Restored printers and settings from Cura Backup.

trying to slice a model and the progress bar goes all the way to the right and then the bar goes blank. Shows it is still slicing but no progress and the slicing never finshes.
Model slices on 4.12.1 correctly.

### Actual results

model never slices and produces g-code,

### Expected results

produce g-code

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

can not attach project file

fvrmr, 2021-12-06:
Hi @moosehaed thank you for your report.
You can share your project file in a .zip here on GitHub.

moosehaed, 2021-12-06:
I get an error whenever I try to attach a file. "is not included in the list".

fvrmr, 2021-12-07:
Attaching a zip file should be working, 3mf is indeed not working unfortunately.

moosehaed, 2021-12-07:
> Attaching a zip file should be working, 3mf is indeed not working unfortunately.

Ive tried chrome and incognito mode in chrome. I can no drag the zip file from my system to this box. Is there another way to attach the zip file?

fvrmr, 2021-12-10:
You can use WeTransfer, Google drive or Dropbox for a link you can share here.

moosehaed, 2021-12-14:
https://drive.google.com/file/d/1WkrdbcBqsw1ExxQQ6vTZQ-hd2Ou9ClPI/view?usp=sharing

Ghostkeeper, 2021-12-17:
That file is protected and requires a log-in to access. I think you need to configure it to be public or get a public sharing link from Google Drive. We can't access it right now.

closed 2021-12-31 by no-response[bot]