Ultimaker/Cura #11019: Gradual infill ignores steps

doppelgrau, 2021-12-06:
### Application Version

4.12.1

### Platform

Linux/Ubuntu

### Printer

Tenlog TL-D3 Pro

### Reproduction steps

Slice the parts mentioned in #11016 with the same profile. just for better visulaization set infill to 70% and steps to 5

### Actual results

On a layer before some part has a top layer the whole infill is printed with 70% infill, ignoring that most area has way lesser infill before so 35% or even 17.5% infill => bad results. After that dense layer, it is printed sparse again.
Only the area below the part that had the surface that infill gradually become denser.

[image]
[image]

### Expected results

Area with no surface should keep the spare infill. But as I learned in closing #11016 this is now expected behavior. but in that case the infill steps should not be ignored.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Still believe the areas that have way more Z to print should keep the spare infill.

fvrmr, 2021-12-07:
Hi @doppelgrau thank you for your bug report. 
I have discussed this with the team and we decided to disable gradual infill when using gyroid as infill pattern. This might not be the solution you would like. 
Devs see CURA-8735

doppelgrau, 2021-12-07:
Thank you (and the team) for looking in that issue.
For the application I heavily use gradual infill the properties of the infill are no so important, was just my "normal one" for infill. So perfectly fine with me. Didn't think it depends on the infill pattern used.

fvrmr, 2021-12-07:
I will reopen this issue. We will close the issue if the implementation is done!