Ultimaker/Cura #11021: When material type is ABS, build Plate Temperature is always fixed to 70degC.

Guswib, 2021-12-06:
### Application Version

4.12.1

### Platform

Windows 10

### Printer

MP 10 mini

### Reproduction steps

I switched between different materials using the menu system. 

### Actual results

When material type is ABS, **build Plate Temperature** is always fixed to 70degC. 
This is the results, even though in the preference of the specific material, different build plate temperatures are given. 

### Expected results

When switching between different materials and materials types, the **build Plate Temperature** changes accordingly given the different material settings. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

The above information should be enough to reproduce the issue.

fieldOfView, 2021-12-06:
> The above information should be enough to reproduce the issue.

Actually, even an empty project is immensely useful to see how you have configured your printer in Cura and what profiles affect the settings that are used.

So, without the benefit of having that information at my fingertips, I'll wager a guess that the printer definition you use has quality profiles that override what is set in the material profile. In the "stack" of profiles that Cura uses, what is set in a quality profile is deemed higher priority than what is set in the material.

Guswib, 2021-12-07:
Dear Aldo,
You are correct. Some hidden settings of the printer makes this, as using another printer does result in the expected temperatures settings. 
I find it strange that this over-ruling is not transparent to the user, and that there is no way one can see this setting of the printer under the printer settings. Would be nice to just see under some printer settings tab.

closed 2021-12-07 by Guswib

Ghostkeeper, 2021-12-17:
The profile system in Cura kind of assumes that the profile authors know what they are doing. They allow quality profiles to override these settings from a material, but if they do it will really override the setting from the material profile.
Maybe it's too flexible.

I think showing this information in the interface does make the interface way more complex though, and is not really appropriate. There are plug-ins such as the [God Mode](https://github.com/sedwards2009/cura-god-mode-plugin) plug-in that make this information visible for those that really need this information.

Ghostkeeper, 2021-12-22:
> Here, I would not mind having the option to set few more settings that are material dependent(print speed for example), especially if I create my own material.

There is also the Material Settings plug-in for that, which allows you to store way more settings in those material profiles. Just be aware that some may be overridden by the quality profile.