Ultimaker/Cura #11024: No operation with my Anet A8 printer

roljol, 2021-12-06:
### Application Version

4.12.1

### Platform

Windows 10

### Printer

Anet A8

### Reproduction steps

Recently installed Cura version 4.12.1 
For the printer my choice is Anet A8

### Actual results

Slicing Ok.
Printer is not recognized.
There are connection attempts.
Printing works with Simplify3D.
With Simplify3D there is a gadget button for the connection to the Com port (115200 bauds) that I have to use. 

Information: the Anet A8 printer is recognized with Cura V3.2.1 

### Expected results

Every 3 seconds the printer is activated and then quickly disconnects.
It is not proposed "Print" but only "Save to disk". 

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/7663940/cura.log)

fvrmr, 2021-12-07:
Hi @roljol thank you for your report. 
We don't maintain USB printing anymore since our printers doesn't use USB printing. And we can't test this for every printer. 
You could check if the USB printing plugin is disabled. 
Go to marketplace - installed and scroll down to find the USB printing plugin.

I will close this issue since this is unfortunately not something we will fix. I hope you understand.

roljol, 2021-12-10:
Thank you for taking the time to read my comment.
If I find a solution, I'll share it with you.