Ultimaker/Cura #11025: 'Save', 'Save As', and 'Delete' Buttons or Text for Managing Profiles

Garryb123, 2021-12-06:
### Is your feature request related to a problem?

Using the drop-down UI element to 'Create', 'Update', and 'Discard' profiles requires too many clicks and mouse movements and can be greatly simplified.

### Describe the solution you'd like

I propose 'Save', 'Save As', and 'Delete' buttons or text for managing profiles. In addition to the effort and time savings (fewer mouse clicks and moves are much more efficient) using 'Save', 'Save As', and 'Delete' would conform with commonly understood industry standards.

Of course the current naming convention could be retained if switching naming was too jarring. The name used isn't the critical aspect but rather the function. As well, the current drop-down structure could be retained for those who like the status quo; _the current and new systems can co-exist easily._

Also, as in the image the word 'Profile' in front of the drop-down can be eliminated as it serves little purpose.

[image]

### Describe alternatives you've considered

Continue to click and move the mouse more than needed. :-)

### Affected users and/or printers

All printers and users would be affected (in a good way).

fvrmr, 2021-12-09:
Hi @Garryb123 thank you for your feature request.
I will bring this up with our UX designer. Keep you posted.

Garryb123, 2021-12-09:
> Hi @Garryb123 thank you for your feature request. I will bring this up with our UX designer. Keep you posted.

Thank you @fvrmr for entertaining my request. All the best.

fvrmr, 2021-12-10:
I have discussed it with our UX designer and the team. And we like the idea of easily saving profiles. So we have a ticket on our backlog related to this request. 
Devs see CURA-8750

Garryb123, 2021-12-10:
> I have discussed it with our UX designer and the team. And we like the idea of easily saving profiles. So we have a ticket on our backlog related to this request. Devs see CURA-8750

This is great news! I look forward to your eventual implementation. All the best.