Ultimaker/Cura #11028: External brim or raft for small support trees to aid adhesion

molove, 2021-12-07:
### Is your feature request related to a problem?

I really like tree supports especially from the build plate, often I have a piece that only needs small support trees. I try not to use brims on pieces that don't need them. Unless I use a brim, often small support trees don't stick to the print bed.

For instance something like this would probably not stick to the bed

[image]

### Describe the solution you'd like

I'm not sure if this has been asked for before but I would like the option to add an external brim or raft to support trees only to aid bed adhesion and not to the piece as well. Or perhaps just to support trees whose bases are less than a certain area.

### Describe alternatives you've considered

I've tried the internal brim for supports but it doesn't help. Using a brim for the model does work but isn't an ideal solution as it adds unecessary clean up to the model.

### Affected users and/or printers

I think it would be beneficial to everyone.

fvrmr, 2021-12-09:
Hi @molove thank you for your feature request. 
I have discussed this with an colleague and we believe this a edge case and this is not a feature with high priority for us to add right now. That's why I will defer this request.

However there is a workaround that can help you. You can use the `brim distance` setting for the normal brim, this will create a gap between the brim and the model and this will not affect the brim for the support. So the brim will be attached to the support but not the model.

Phusck, 2023-03-06:
WOuld really like to see this

onolox, 2023-03-26:
This feature is REALLY needed, it's not a "edge case".
Trees really need a bigger base to stay in the bed.

CWempe, 2023-08-17:
I too think it would make sense to add a brim option for supports only.

Until then, I add a little info to the workaround, because only setting the "Brim Distance" is not the best/cleanest option for me.

Without brim:
[image]

With brim:
[image]

With 2 mm brim distance:
[image]

With 2 mm "brim distance" and 0 mm "Brim Width": 👎
[image]

With 2 mm "brim distance" and 1 "Brim Line Count": 👍
[image]

So you cannot get rid of the last Brim Line, but you can set a bigger distance.
Unfortunately the distance is valid for support trees. too.-
But at least there is a brim inside of the tree.

klwemu, 2023-10-29:
Tree Support does not stick enough. And brim distance is not a working solution. 
Tree support with brim is in PrusaSlicer and OrcaSlicer a pretty cool default feature.

Quedale, 2023-12-15:
I also agree that such feature would be really useful.
The higher a tree support gets, without touching the model with a branch, it becomes really weak.
Tall(ish) trees only need a very slight drag from the nozzle/filament to pull the entire tree from the plate.

The only workaround I found is to use a raft, which may defeat the purpose of using tree support in the first place depending on the model.

adamsthws, 2023-12-18:
This is something I'd love to see in Cura too!

njsokalski, 2024-05-09:
I agree that this would be extremely useful and advantageous, I have had tons of Tree Supports collapse because they got too tall without enough adhesion, and am yet to find a good solution.

closed 2025-05-23 by github-actions[bot]