Ultimaker/Cura #11029: Crash on opening STL

pvfromtours, 2021-12-07:
### Application Version

4.12.1

### Platform

Windows 10

### Printer

Ultimaker 3

### Reproduction steps

Try to open STL file on clicking on open icon or 

### Actual results

Crash when open a STL file.
Do not happen with 3MF files.

### Expected results

open file

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/7668611/cura.log)

fvrmr, 2021-12-07:
Hi @pvfromtours thank you for your report. 
Does this happen with every stl file? Do you have Chitubox installed?

pvfromtours, 2021-12-07:
Hi fvrmr
Yes with all stl on 4.12, 4.11, 4.10, I reinstalled 4.8 and it now works.
Yes I have Chitubox 1.9.0

fvrmr, 2021-12-10:
If you have Chitubox installed this is a duplicate of https://github.com/Ultimaker/Cura/issues/9644
See this support article what you can do to fix this: https://support.ultimaker.com/hc/en-us/articles/4402996140946-Ultimaker-Cura-crashes-when-opening-any-file

I will close this issue since it is a duplicate.

closed 2021-12-10 by fvrmr

pvfromtours, 2021-12-10:
Thanks a lot Fenn.