Ultimaker/Cura #11031: slicing error

philchasebikenerd, 2021-12-07:
### Application Version

4.12.1

### Platform

windows 10

### Printer

creality cr10 mini

### Reproduction steps

i downloaded the new version of cura 
I imported a model on cura for slicing
I selected my desired printing settings
I sliced the model with my desired settings
I selected the preview option to verify if the print is ready to print

### Actual results

the slicing did not include infill or top layers without giving me any error notice.
this bug is happening with all previous models I already printed.
[image]

### Expected results

That my model is sliced corectly with no missing elements.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura bug.zip](https://github.com/Ultimaker/Cura/files/7670149/cura.bug.zip)

GregValiant, 2021-12-08:
Turn off "Spiralize Outer Contour".

fvrmr, 2021-12-09:
Hi @philchasebikenerd thank you for your report. 
Like @GregValiant mentioned disable Spiralize Outer Contour will solve your issue. 
I will close this issue, but please reach out if you have more questions.

closed 2021-12-09 by fvrmr