Ultimaker/Cura #11034: Filename of stl persist even after clearing the

ericnoel, 2021-12-08:
### Application Version

4.12.1

### Platform

macOS 12.0.1

### Printer

Ender 3 V2

### Reproduction steps

1. Newly opened Cura with used from last closed profile
2. Drag an stl file to Cura layer (in Prepare stage)
3. File name of the STL is displayed in title bar
4. Slice 
5. Save
6. Open 
6. Undo two (2) times to remove/clear the stl object from the screen
7. Drag another STL file to Cura layer (in Prepare stage)
8. File name of the STL is displayed in title bar
9. Slice 
10. Save
11. Open 
12. Undo two (2) times to remove the stl object from the screen

### Actual results

13. Notice that the Filename from previous stl still persist/ not cleared

### Expected results

Filename should have been cleared and acquire the filename from the newly dragged object

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Even if multiple undos, filename still not cleared

fvrmr, 2021-12-09:
Hi @ericnoel thank you for your report.
If you use clear build plate the name will be updated. 
I will bring it up with the team. Keep you posted!

fvrmr, 2021-12-10:
I have discussed this with the team and we decided to defer this issue since it has not that much priority for us to fix right now. And using Right click - clear build plate or cmd D will remove the file name. I hope you understand.