Ultimaker/Cura #11035: Cura closes when opening file explorer

qwert95, 2021-12-08:
### Application Version

4.12.1

### Platform

Windows 10, AMD Vega 56 GPU

### Printer

Ender 3

### Reproduction steps

- Open cura
- Click on the top left icon to open the file explorer

### Actual results

The file explorer shows for a couple of seconds, then Cura closes.

### Expected results

File explorer opens

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/7676091/cura.log)

Additional info: I can still open STL files in Cura if I drag them to Cura's window from Windows 10's own file explorer.

nallath, 2021-12-08:
Please check https://support.ultimaker.com/hc/en-us/articles/4402996140946-Ultimaker-Cura-crashes-when-opening-any-file

qwert95, 2021-12-08:
It worked. Thank you.

closed 2021-12-08 by qwert95