Ultimaker/Cura #11038: Chrome OS supported

cloudy-fem, 2021-12-08:
### Is your feature request related to a problem?

Hey, it would be really handy if Cura was compatible with Chrome OS. So that I can use it during class. Recently, our Imac's and windows computers have not been working as they should. Only one turns on correctly, but it refuses to run cura as it closes/quits unexpectedly when trying to open. Maybe check that out? (will issue a bug report for this) I would love to be able to make prints on my Chromebook as my Acer computer at home does not have an SD slot.

### Describe the solution you'd like

I believe this will solve most troubles people may have because normally, a Chromebook with Chrome OS runs better than any apple or windows computer. This would be rather nice to do.

### Describe alternatives you've considered

I use TinkerCad to build my projects then use Cura to make the Obj. Stl. files a Gcode. file.

### Affected users and/or printers

The only printer that I use with the Ultimaker cura seems like it does not get sold anymore. The Ultimaker 2. I'd like to be able to use this a bit better but then again, it seems like these do not get made/manufactured anymore.

smartavionics, 2021-12-08:
Hi, it's already possible to run Cura on a chromebook. I don't know about the UM releases but my releases do run. They're not speedy but they do work. The most recent release for Intel processors is https://github.com/smartavionics/Cura/releases/download/20211119/Cura-mb-master-x86_64-20211119.AppImage

This AppImage requires the chromebook to be using bullseye due to a library dependency but my earlier releases run on chromebooks that use buster.

Here's the proof!

[image]
[image]

smartavionics, 2021-12-08:
[image]