Ultimaker/Cura #11040: Slicing failed

AlfredAn, 2021-12-08:
### Application Version

Arachne_engine_beta_2

### Platform

Windows 10, GTX 1060 6GB

### Printer

Creality Ender 3

### Reproduction steps

1. Open the provided project file in Cura.
2. Try to slice it.

### Actual results

An error message appears saying that slicing failed. See the attached image for the full text.

[image]

### Expected results

The model slices as normal.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[bug_report.zip](https://github.com/Ultimaker/Cura/files/7679356/bug_report.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/7679365/cura.log)

fvrmr, 2021-12-09:
Hi @AlfredAn thank you for your bug report. 
If you move your objects a bit further apart, you can slice your objects. 
But that's strange because that would suggest this is related to the `Print Sequence: One at a Time`. Changing this to `All at Once` will also show that the slicing failed.
I will bring this up with the team. Keep you posted!

fvrmr, 2021-12-10:
I have discussed it with the team and we decided to defer this bug report. We believe this is a edge case, since if you move your models a bit further apart it will slice. I hope you understand.

closed 2023-06-09 by github-actions[bot]