Ultimaker/Cura #11041: First M117 Placement

Trek2120, 2021-12-09:
### Application Version

4.12.1

### Platform

Windows 10

### Printer

Creality CR-10S

### Reproduction steps

Generated Gcode file

### Actual results

First M117 command is rendered ineffective because if the other commands that update the printer's screen.

### Expected results

I always move the first M117 command to just below the ;LAYER:0 line of code.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Thank you
[cura.log](https://github.com/Ultimaker/Cura/files/7681322/cura.log)

fvrmr, 2021-12-09:
Hi @Trek2120 thank you for your report. 
Are you using one of the post processing scripts `Display Progress On LCD` or `Display Filename And Layer On LCD`?
Could you share a project file (file - save project)?

Trek2120, 2021-12-31:
Sorry for the delay.  I’ve been out of town. I’m using the “Post Processing Scripts”.  The first M117 command is line three of the Gcode file.  Line 22 starts the code I have for the “Start G-code” for my printer. I’ve been moving the M117 line down below the “;LAYER:0” line so that it will display.  I recently discovered that I can change the frequency to “Every 15 seconds”, so it hasn’t been an issue anyway.  Just thought I’d point it out. Thanks! Doug ;FLAVOR:MarlinM73 P0M117 Time Left 0h32m12s   ;TIME:1932;Filament used: 1.08512m;Layer height: 0.2;MINX:92;MINY:112.5;MINZ:0.2;MAXX:146.8;MAXY:132.3;MAXZ:5;POSTPROCESSED;Generated with Cura_SteamEngine 4.12.0M140 S60M105M190 S60M104 S205M105M109 S205M82 ;absolute extrusion mode; Ender 3 Custom Start G-codeG92 E0 ; Reset ExtruderG28 ; Home all axesG1 Z2 F3000 ; Move Z Axis up little to prevent scratching of Heat BedG1 X2 Y20 Z0.3 F5000.0 ; Move to start positionG1 X2 Y200.0 Z0.3 F1500.0 E15 ; Draw the first lineG1 X4 Y200.0 Z0.3 F5000.0 ; Move to side a littleG1 X4 Y20 Z0.3 F1500.0 E30 ; Draw the second lineG92 E0 ; Reset ExtruderG1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat BedG1 X6 Y20 Z0.3 F5000.0 ; Move over to prevent blob squishG92 E0G92 E0G1 F1500 E-6.5;LAYER_COUNT:25;LAYER:0M107;MESH:LaserCalibration.stlG0 F6000 X116.4 Y129.9 Z0.2 From: FenneSent: Thursday, December 9, 2021 8:56 AMTo: Ultimaker/CuraCc: Trek2120; MentionSubject: Re: [Ultimaker/Cura] First M117 Placement (Issue #11041) Hi @Trek2120 thank you for your report.Are you using one of the post processing scripts Display Progress On LCD or Display Filename And Layer On LCD?Could you share a project file (file - save project)?