Ultimaker/Cura #11042: 4.12.1 Crash when restoring online backup

rdurham, 2021-12-09:
### Application Version

4.12.1

### Platform

Windows 10 1809

### Printer

Custom FFF Printer

### Reproduction steps

1. Clean uninstall all Cura files and registry. Install 4.12.1
3. Add Custom FFF Printer with all defaults, load calibration cube, test functionality, quit app
4. Load app, sign into marketplace, load calibration cube, test functionality, quit app
5. Load app, restore last online backup from Ultimaker account

### Actual results

Last online backup loads and functionality is normal.

### Expected results

Last online backup does not load and app appears to crash.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

I had this happen 2 or 3 times before I did the clean uninstall, all with the same results.

I think it was happening in 4.12.0 also.

[cura.log](https://github.com/Ultimaker/Cura/files/7682235/cura.log)

rdurham, 2021-12-09:
Same thing in 4.11.0: [cura.log](https://github.com/Ultimaker/Cura/files/7682328/cura.log)

I will go back to 4.12.1 and see what all I can do without restoring the online backups.

The online backups say 4.6.1 by the way.

rdurham, 2021-12-10:
I realized that after the crash, my profiles are available, and the app is stable. I'm happy.

For the devs, I was watching filesystem changes during the online backup restore:

[image]

[stdout.log](https://github.com/Ultimaker/Cura/files/7689785/stdout.log) at the top is the last file to be modifed before the crash. The plugin files below were modified earlier.

fvrmr, 2021-12-10:
Hi @rdurham thank you for your report. 
We experienced some issues ourselves as well with back up. We have a ticket to investigate what is happening. 
Devs see CURA-8751

djlightflighter, 2021-12-12:
> ### Application Version
> 4.12.1
> 
> ### Platform
> Windows 10 1809
> 
> ### Printer
> Custom FFF Printer
> 
> ### Reproduction steps
> 1. Clean uninstall all Cura files and registry. Install 4.12.1
> 2. Add Custom FFF Printer with all defaults, load calibration cube, test functionality, quit app
> 3. Load app, sign into marketplace, load calibration cube, test functionality, quit app
> 4. Load app, restore last online backup from Ultimaker account
> 
> ### Actual results
> Last online backup does not load and app appears to crash.
> 
> ### Expected results
> Last online backup loads and functionality is normal.
> 
> ### Checklist of files to include
> * [x]  Log file
> * [ ]  Project file
> 
> ### Additional information & file uploads
> I had this happen 2 or 3 times before I did the clean uninstall, all with the same results.
> 
> I think it was happening in 4.12.0 also.
> 
> [cura.log](https://github.com/Ultimaker/Cura/files/7682235/cura.log)
> 
> Edit: Corrected mistake in results

I had the same issue upgrading from 4.11 and boom my plugin load and that's all she wrote  even tried fresh install and same thing plug ins come in and it breaks

nallath, 2021-12-15:
I've tried the following:
1. Create a backup in an older Cura version (4.8)
2. Load Cura 4.12.1
3. Remove the profiles that got upgraded (So i could see that the backup worked)
4. Restore the backup (Cura will automatically close after this!)
5. Restart Cura
6. See if the profiles are there

The profiles were correctly restored.